iNavFlight/inav #9610: MAMBAF722_I2C GPS+Magnetometer don't work together

Vins1417, 2024-01-01:
## Current Behavior
On the Mamba F722 35A MK1, target MAMBAF722_I2C, the gps connects to the uart4 and SCL & SDA pads.

After flashing firmware 7.0.0 the barometer and magnetometer are recognized without problem. (Note that in the basic configuration the UART4 port is checked MSP).

When you configure UART4 as GPS, the MSP check mark disappears and the barometer and magnetometer are no longer recognized and appear in red.

## Steps to Reproduce

1. Flash to firmware 7.0.0 -> the barometer and magnetometer works fine.
2. Configure UART4 as GPS, save and reboot
3. The barometer and magnetometer are no longer recognized and appear in red

## Expected behavior
Configure UART4 as GPS save and reboot and magicaly the barometer and magnetometer works fine.

---

# version
# INAV/MAMBAF722_I2C 7.0.0 Dec  5 2023 / 10:21:16 (895a4f31)
# GCC-10.3.1 20210824 (release)
[image]
[image]
[image]

mmosca, 2024-01-09:
Did you test with usb only, or did you test with the battery connected? Is the gps actually powered?

sensei-hacker, 2024-01-10:
The I2C board has the built-in Bluetooth on UART4. That's why it's set for MSP.
You could try connecting GPS to UART5.

Vins1417, 2024-01-10:
> Did you test with usb only, or did you test with the battery connected? Is the gps actually powered?

hoo it works perfectly with the battery!!  Thanks a lot !!  but it's strange because without the GPS the barometer was fine without the battery.

Vins1417, 2024-01-10:
> The I2C board has the built-in Bluetooth on UART4. That's why it's set for MSP. You could try connecting GPS to UART5.

are you sure ??  because I never knew that there was bluetooth on this card.. and in the doc the gps is on uart4... you weren't confusing it with the f722 APP stack??

mmosca, 2024-01-10:
> 
> hoo it works perfectly with the battery!! Thanks a lot !! but it's strange because without the GPS the barometer was fine without the battery.

Hard to figure out. But the unpowered magnetometer may have caused issues in the i2c bus. I will close this, since connecting the battery fixes the issue.