iNavFlight/inav #9612: SpeedybeeF405V3 target has unreachable S9 servo output

artempolikarpov, 2024-01-01:
**Summary:**
SpeedybeeF405V3 FC has S9 servo output pin. 
[image]
If number of added servo is 5, there is PWN INIT ERROR on OSD and arming is disabled.
[image]

**Steps to reproduce:**
1. Add 5 servos on Mixer tab and Save
2. Go to CLI tab and type _STATUS_

**Actual result:**
Arming is disabled
_PWM output init error: Not enough servo outputs/timers_ - message in CLI tab is displayed

`# status`
INAV/SPEEDYBEEF405V3 7.0.0 Dec  5 2023 / 11:05:24 (895a4f31)
GCC-10.3.1 20210824 (release)
System Uptime: 63 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=BMI270, ACC=BMI270, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1920
I2C Errors: 0, config size: 9513, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 507, PID rate: 1972, RX rate: 9, System rate: 9
**Arming disabled flags: RX CLI PWMOUT**
OSD: MAX7456 [30 x 16]
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz
**PWM output init error: Not enough servo outputs/timers**

sensei-hacker, 2024-01-01:
You can try this file. I don't know for sure if it will work, so please let us know the results of your test.
https://drive.google.com/file/d/1cVT0FKUl1v_e3op8rYJvoRbED5DSBLNh/view?usp=sharing

Also, this issue should actually be in the INAV repo rather than INAV Configurator, so I have moved it.

DzikuVx, 2024-01-02:
Fixed in https://github.com/iNavFlight/inav/pull/9614

PR #9614 (2024-01-05): Allow S9 on SpeedyBee F405 V3
fixes #9612
closed 2024-01-02 by DzikuVx