iNavFlight/inav #9615: Inav 7.0 autolevel issue

fatbaldmen, 2024-01-02:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV 

## Current Behavior

1. Attempted to fly inav 7 on different Flight controllers. All producing the same issue
2. Word of mouth from some of the other inav testers that this is also happening to them.
3.Created a post in the inav fixed wing Facebook group, asking if anyone else has had this issue. 
4.

## Expected behavior

---

- FC Board name and vendor: Matek F411-wte, SpeedyBee F405 wing 
- INAV version string: Inav 7.0 stable, with inav 7.01 configurator

breadoven, 2024-01-04:
What value do you have for `fw_level_pitch_gain `?

There was bug with auto level which theoretically would have prevented it working properly if at all. It's possible it did work just very slowly. This bug was fixed in 7.0 which possibly would have made it a lot more responsive than before. Reducing the above setting should settle the response if it's causing problems.

fatbaldmen, 2024-01-04:
@breadoven 

FW_LEVEL_PITCH_GAIN = 5.00

closed 2024-01-22 by breadoven