iNavFlight/inav #9619: Build process exits with error "ERROR [6/8] RUN if [ -n "1000" ]; then USER inav; fi"

Lebensgefahr, 2024-01-04:
## Steps to Reproduce
1. git clone https://github.com/iNavFlight/inav.git
2. ./build.sh FLYWOOF7DUAL

## Expected behavior
Builded firmware file

## Suggested solution(s)
I suggest to rollback changes added by commit described below for Dockerfile file.

## Additional context
Building process was broken by this changes.

`commit ae6b6aa7c0ea070cc5ebf452cfba007039c0a96c
Author: Roman Lut <11955117+RomanLut@users.noreply.github.com>
Date:   Tue Oct 3 21:32:13 2023 +0200

    docker build: do not run user commands

diff --git a/Dockerfile b/Dockerfile
index 9c816b0c1..e0a394951 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -4,15 +4,15 @@ ARG USER_ID
 ARG GROUP_ID
 ENV DEBIAN_FRONTEND noninteractive
 
-RUN apt-get update && apt-get install -y git cmake make ruby gcc python3 python3-pip gcc-arm-none-eabi
+RUN apt-get update && apt-get install -y git cmake make ruby gcc python3 python3-pip gcc-arm-none-eabi ninja-build
 
 RUN pip install pyyaml
 
 # if either of these are already set the same as the user's machine, leave them be and ignore the error
-RUN addgroup --gid $GROUP_ID inav; exit 0;
-RUN adduser --disabled-password --gecos '' --uid $USER_ID --gid $GROUP_ID inav; exit 0;
+RUN if [ -n "$USER_ID" ]; then RUN addgroup --gid $GROUP_ID inav; exit 0; fi
+RUN if [ -n "$USER_ID" ]; then  RUN adduser --disabled-password --gecos '' --uid $USER_ID --gid $GROUP_ID inav; exit 0; fi
 
-USER inav
+RUN if [ -n "$USER_ID" ]; then USER inav; fi
`
I can't understand what did author trying to do. I think it's just a miss understanding of docker build process. Or maybe he is trying to run docker build as root for example.
The first two RUN strings can be fixed by removing the second RUN word so user will be added if build arg USER_ID is specified.
But the last one RUN can't be made at all. Dockerfile instructions can't be used this way. Installing needed software for debugging (GDB) isn't a good decision for all. If somebody needs to debug he can just build his own inav-build image with this software. It's just increase building time, image volume and internet traffic.

---

- FC Board name and vendor:
- INAV version string:

jschluchter, 2024-01-21:
I can second that this does indeed fail. Running OSX Sonoma 14.2.1 on M1 Pro.

RomanLut, 2024-02-06:
1. Ok I can add:
```RUN if [ "$GDB" = "yes" ]; then apt-get install -y gdb; fi```

so debugger will be installed when process is run with GDB=yes parameter only:
```docker build --build-arg GDB=yes -t inav-build  .``` 

2. On Windows/WSL 2 mounted /src folder is writeable for root user only. I have to run build under root user:

```docker run --rm -it -u root -v <PATH_TO_REPO>:/src inav-build <TARGET>```

The following lines should not be run under Docker on Windows (process is started with with above line):

```
RUN addgroup --gid $GROUP_ID inav; exit 0;
RUN adduser --disabled-password --gecos '' --uid $USER_ID --gid $GROUP_ID inav; exit 0;
USER inav
```

Could you propose solution with correct syntax?

Lebensgefahr, 2024-02-06:
> 1. Ok I can add:
>        `RUN if [ "$GDB" = "yes" ]; then apt-get install -y gdb; fi`
>
>
> so debugger will be installed when process is run with GDB=yes parameter only: `docker build --build-arg [...]
You don't need to edit Dockerfile in a repository just add or remove in a local copy. It is doing it's job as it was designed. 
USER Dockerfile instruction can't be used with any conditions statement. That means you can't make workaround by editing Dockerfile. All other edited strings are meaningless because if the root user is already exists in the source image (FROM ubuntu:jammy) that will not cause an error in adduser/addgroup commands and inav user will have id:0 and gid:0.
I suggest you to use build.sh for building image. Of course you need to install something like ubuntu in your windows system.
For example: https://stackoverflow.com/questions/52512026/is-it-possible-install-ubuntu-in-windows-10-wsl-without-microsoft-store
Something like this:
```
# could do it like this (from PowerShell):
# go into some folder into which you want the file to be downloaded
cd <somefolder>

# download Ubuntu 20.04
Invoke-WebRequest -Uri https://aka.ms/wslubuntu2004 -OutFile Ubuntu.appx -UseBasicParsing

# install downloaded *.appx file
Add-AppxPackage .\Ubuntu.appx
```
After that Ctrl+Esc type Ubuntu and run it. It will prompt you to add user. 
After that you can run bash right from cmd/Win+R/powershell.

RomanLut, 2024-02-06:
Installing linux is not a solution, and editing dockerfile locally either.
The goals is to make Dockerfile usefull on all host systems.

Please check if fixed Dockerfile works fine for you: https://github.com/iNavFlight/inav/pull/9702

Lebensgefahr, 2024-02-06:
> Installing linux is not a solution, and editing dockerfile locally either. The goals is to make Dockerfile usefull on all host systems.
> 
> Please check if fixed Dockerfile works fine for you: #9702

It is already prepared for PROPER using in any systems. By the topic error I can guess you are beginner in docker usage. Consequently you are making other common errors like file permissions. Perhaps it’s worth understanding better how to use it and not breaking what already works?
Yeah it sounds toxic but I ate my shorts while trying to build new firmware image "quickly".
Now it is building but please make a note for yourself.

RomanLut, 2024-02-06:
1) Yes, I am not expert in docker

2)  As I wrote, image should be run with "root" user on Windows host.
The line "USER inav" makes it impossible to use Dockerfile on Windows.

[image]

 Basically I reverted all changes except:
```
ARG INAV_IMAGE_USER=inav
USER $INAV_IMAGE_USER
```
which replace:
```
USER inav
```

which should make everyone happy.

Lebensgefahr, 2024-02-06:
The proper way is to run ./build.sh with suitable target. 
Just open it and you will see that USER_ID and GROUP_ID are passing to docker build command as build args.
So if I am run your docker build command before your changes I will get the same error. And this is not windows issue. 
[image]
This happened because I didn't specify this args to docker build and user inav was not created.
Now just run docker build properly:
`docker build -t inav-build --build-arg USER_ID=1000 --build-arg GROUP_ID=1000 .`
[image]
USER_ID and GROUP_ID should not be equal to already existent users.
This mechanism in ./build.sh is a common way to avoid file permissions issue. This trick makes current user the owner of files created by user in the container.
In fact the command 
`RUN addgroup --gid $GROUP_ID inav; exit 0`
should not be finished with `exit 0` I don't know why it was added. It's only hide an real error of adduser/addgroup.
[image]

RomanLut, 2024-02-06:
Thanks for the hint, passing (dummy) 1000 as user & group ids prevents image building from failing. User  setup does nothing on Windows Host anyway.

I reverted:
```
ARG INAV_IMAGE_USER=inav
USER $INAV_IMAGE_USER
```
and updated documentation.

closed 2024-02-08 by DzikuVx