iNavFlight/inav #9629: SRXL2 receiver not working on UART 5

pgodoyc, 2024-01-11:
## Current Behavior
SRXL2 receiver connected to UART 5 stopped working after upgrading to INAV 7. 

INAV/AOCODARCF4V2 6.1.1: receiver connected to UART 5 works ok. 
INAV/AOCODARCF4V2 7.0.0: receiver connected to UART 5 does not work. 

Downgraded to 6.1.1 and receiver functionality was restored. So, it is not wiring. 

## Steps to Reproduce

1. Upgrade FW to 7.0.0
2. Connect SRXL2 receiver to UART 5
3.
4.

## Expected behavior
Enable receiver operation. 

## Suggested solution(s)
Enable receiver operation. 

## Additional context

Working:
# version
# INAV/AOCODARCF4V2 6.1.1 Jun 13 2023 / 09:56:20 (42ced253)
# GCC-10.2.1 20201103 (release)

Not working: 
# version
# INAV/AOCODARCF4V2 7.0.0 Dec  5 2023 / 09:30:52 (895a4f31)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-22:
7.0 had multiple changes in the DMA and timer code to solve different issues. 
It might be that this is why it does not work anymore on UART5 if it has no DMA. Can't say for sure. 

Anyway, based on the documentation of the FC, the receiver is supposed tobe connected to UART2. Usually there are reasons why FC manufacturer advise specific serial ports for receivers or other hardware. Try UART 2 that should work.

closed 2024-03-22 by b14ckyy

pgodoyc, 2024-03-22:
I would suggest to eliminate UART 5 from the list of supported ports for this FC. 

A fix is not required - as you can use a different port - but it would be good to be clarified in the documentation. 

Thanks for your help and for all the good work you put to develop this.

b14ckyy, 2024-03-22:
UART 5 can still be used for other things like Digital OSD or GPS. When it comes to wiring, the FC Manufacturer documentation should always be number one source. Some FCs have very specific quirks that INAV can't correct for all of them. If you still have issues with UART 2, ping me here.