iNavFlight/inav #9634: In Motor Test, the motor don't spin if the command is lower than configured 'Motors IDLE power'

her01n, 2024-01-12:
## Current Behavior

The motor does NOT spin.

## Steps to Reproduce

1. Connect the configurator
2. Select Outputs tab
3. Configure Motors IDLE power to 15%
4. Enable 'I understand the risks, ...' checkbox to enable the motor test mode
5. Move any motor slider to a value bigger than minimum required for the motor to spin, but below 'Motors IDLE power', for example 10%

## Expected behavior

The motor would spin.

## Additional context

With MULTISHOT, the motors would spin with the described input. Only DSHOT is affected.

## Dump

https://pastebin.com/JbkvepNe

- FC Board name and vendor:
speedybee f403 v3
- ESC:
speedybee  BL32 50A V3 - not the one that comes with f403 in stack
- INAV version string:
# version
# INAV/SPEEDYBEEF405V3 7.0.0 Dec  5 2023 / 11:05:24 (895a4f31)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-22:
I think this is work as intended. The fact that it acts differently between protocols might be an issue. Maybe worth a look.

pkojder, 2024-05-15:
Not working for me 
I got Mateksys F405TE_SD on 5" Quad with Inav 7.1.1. After arming motors doesn't spin at all. Also when I go to Motor test mode after raising a bar up motors are not spining

closed 2026-09-27 by sensei-hacker