iNavFlight/inav #9636: INAV 7 not logging yaw set point or rc command

bw1129, 2024-01-13:
I notice that yaw set point and yaw RC command are not being logging in INAV 7. Tried logfiles from serveral different sources. It just shows a string of zeros. roll and pitch are fine. here is a screenshot of an example, zoomed 500%. yaw (YELLOW) is perfectly flat with no deviation from 0.
[image]

DzikuVx, 2024-02-02:
Just quickly checked and everthing looks fine over here 
[image]

Maybe you could share more details @bw1129

b14ckyy, 2024-03-21:
Any input on that or is this obsolete?

bw1129, 2024-03-21:
It seems fine on some FCs so for now I assume it was an SD card logging issue.

closed 2024-03-21 by b14ckyy