iNavFlight/inav #9637: Autolevel trim value jumping from setpoint to 15 degrees when switching to horizon.

a1axx, 2024-01-13:
Inav 7.0
Matek F722 HD

With autolevel trim in OSD, you can see the trim value jump from X.X to 15.0 when switching from Cruise mode to Horizon + Autolevel (on same channel 5 PWM).

I first witnessed this when flying along in horizon, I confirmed autolevel trim was changing in flight and it steadied off at 4.7 degrees.

When flying in cruise then switching back to horizon with autolevel ~ 10 -20 seconds the trim was jumping to 10-11 degrees. When levelled off back to ~4.5 degrees, switched back into cruise and flew around for 10-15minutes, then upon switching back to horizon with autolevel the trim was maxed out at 15 degrees.

As I was on a return leg I had the throttle reduced from 45% to 35%, this meant to pitch up caused by the 15 degrees of level trim caused the plan to have a stall in horizon. I was able to recover this quickly by increasing the throttle to ~70%.

Will confirm if auto level is enabled with cruise, though I don’t think this matters as it does not work in cruise as far as I know.

Steps to Reproduce

1. Setup horizon with auto level on same channel PWM
2. Setup cruise mode
3. Setup level trim + pitch to be displayed in OSD
4. Fly in horizon and confirm successful trim.
5. Switch to cruise mode for 1-2minutes
6. Switch back to horizon
7. Note the change if any in level trim and the increase in pitch angle.

Expected behavior
Level trim value does not change when switching back to horizon.

breadoven, 2024-01-13:
Looks like the bug fix https://github.com/iNavFlight/inav/pull/9393 introduced another bug, probably caused by the time delta not being reset correctly on the first loop following a restart of auto level. It should eventually correct itself given time.

Only short term solution is to change `fw_level_pitch_gain `although it's not clear if increasing or decreasing this would help. Decreasing would make it less responsive but if the time delta bug results in a high value this won't help. so perhaps increasing it would help to make it more responsive reducing the initialising error quicker although this is likely to lead to oscillations so you can't win really. Not sure why it's going to 15 degrees though, the adjustment limit is hard coded to 10 degrees.

Auto level is only active during Angle, Horizon and Course Hold.

https://github.com/iNavFlight/inav/issues/9615 will probably be the related to the same problem.

a1axx, 2024-01-15:
Thanks Breadoven. I am just going to set the level trim to 4.7 and remove auto level for now.

closed 2024-01-22 by breadoven