iNavFlight/inav #9638: Possible gyro readout error

DarthPlasma, 2024-01-13:
Custom BOB57 with Diatone Mamba H743 mark 4

## Current Behavior
After updating to Inav 7 i noticed some Pitch jerk movement, investigating the RAW gyro signal shows some strange occasional full scale spikes (up to 2000 raw value) only on the pitch signal, usually negative. This spike is shaped by the filtering to a longer peak that screws up temporarily the DTERM and hence the motor output, then dissipating in a couple of ms but causing the motors to give a false pitch output that produces noticeable twitch on the motors.

## Steps to Reproduce
It happens everytime I fly since the update, at least 2-5 times each minute.

## Expected behavior
Spikes to be absent on the raw gyro

## Suggested solution(s)
Checking if there is some incompatibility problem with the ICP42688-p readout

- FC Board name and vendor: Diatone Mamba H743 mark 4
- INAV version string: 7.0.0
[image]
[image]
[image]

DarthPlasma, 2024-01-14:
It happens even with the drone armed on ground, props removed 
[image]

mmosca, 2024-02-23:
Can you try 7.1 rc1?
I made some changes to the filtering in the icm42688p gyro and that got rid of a similar issue I had with the same fc.

mmosca, 2024-02-28:
Fixed by #9675 

Feel free to re-open if it still happens after 7.1RC1

closed 2024-02-28 by mmosca

DarthPlasma, 2024-03-04:
The issue seems to be resolved, at least in a quick basement testing 😊

DarthPlasma, 2024-03-06:
Unfortunately the spikes are more rare but not disappeared totally. In 8 minutes arm time i had 8 spikes, and 1 in another 4 minute flight

DarthPlasma, 2024-03-06:
this is the blackbox of the drone armed on ground

 https://we.tl/t-nyEGVo3tnr

DarthPlasma, 2024-03-06:
Ok issue resolved al 100%. The residual spikes were from leds connected on the 5V of the FC. Using a BEC wiped out the last problems