iNavFlight/inav #9642: Kakute H743-Wing with INAV 7.0.0: ledstrip does not work

Torchi, 2024-01-14:
I have got a Holybro Kakute H743-Wing FCB and the problem I report is that I have never seen ON any led of the strip.
Everything else I am testing of the FCB  (servos, peripherals, functions, etc.) is working fine.

I connected a ledstrip (PandaRC LED0539 RGB 5V). 
I loaded at first INAV version 6.1.1 and  it did not work.  Afterwards, I have understood that the ledstrip function is not complete  in 6.1.1 sources.  
Then, I loaded INAV 7.0.0. This version seems to support well the Kakute H743-Wing...  except for the ledstrip, it seems to me, as still I do not see it working  (I have never seen any led turned on).
I have checked, with another model and INAV version,  that the ledstrip itself is perfectly ok.
I think the configuration I use is ok (in any case, I attach the dump). 
The ledstrip pins are connected to:  GND, supply and the M14 pin.  As supply I tried either +5V or  Vx (that is 6V and is dedicated to all servos): neither worked.
[INAV_7.0.0_cli_FUNCUB_20240112_105603.txt](https://github.com/iNavFlight/inav/files/13931701/INAV_7.0.0_cli_FUNCUB_20240112_105603.txt)

b14ckyy, 2024-03-20:
can you please test the Artifact build from here: https://github.com/iNavFlight/inav/actions/runs/7546952104/artifacts/1173380099

this has some fixes for the target definitions that did not match the manufacturer definitions.

Torchi, 2024-03-22:
It does not seem to work, still.
What I did is described in the following.  
I loaded the fw version you sent, variant for H7 WING, i.e.  inav_8.0.0_KAKUTEH7WING_ci-20240116-c19439b.  
My problem to do  a proper led control is I have no v.8 configurator.  
So, from CLI, I have just input (should it be sufficient to light up the first led?):  

feature LED_STRIP
led 0 0,0::C:10

and I connected a 4-led strip, supplied with 5V.  
It did NOT light up.

I attach the diff file.
[release 8 diff.txt](https://github.com/iNavFlight/inav/files/14718200/release.8.diff.txt)

b14ckyy, 2024-03-22:
OK thanks for trying but sounds there is at least some difference. Will do some testing later too. the website for automated snapshot builds is down at the moment to get a 8.0 build and I have no build environment set up. need to work on that :D

Torchi, 2024-03-22:
As conclusion of a day in which I tested the fw version  inav_8.0.0_KAKUTEH7WING....mentioned above.   In the day I have seen the LEDs lighting up 3 or 4 times.   I did, in total, about three hours of testing ie.  several tens of cycles of various configuring, switching on/off, etc.   So , LEDs CAN light up but, for some reason (a gut feeling:  hw initialisation timing?) ,  the successful cases are rare and I could not see any pattern.  I am even thinking that, trying so many times, the previous version 7 could behave same.

b14ckyy, 2024-03-22:
OK I leave it open with the BUG lavel and a dev with the board need to look at it.