iNavFlight/inav #9643: Mixer.md Yaw description does not work with Props-Out.

Phil-MC, 2024-01-14:
A user on the Facebook Group, Oran Tao, raised an issue which  I am passing on. He was dealing with a Props-Out situation.
The wording in Mixer.md which also appears in the Configurator, and is in BOLD below, does not work with Props-Out.
"
YAW
Indicates how much yaw authority this motor imparts to the yaw rate of the aircraft. Range [-1.0, 1.0]. For fixed wing models with more than one motor this weight can be used to setup differential thrust. For fixed wing models with only one motor this is usually set to 0.
> **A positive value means that the motor needs to accelerate for a positive yaw rate request (clockwise yaw seen from the top of the model). A negative value means that the motor needs to decelerate**

## Suggested solution
Either state that this is only true for Props-Out or give that solution as well.
## Additional context
I must say that when I saw what he was highlighting in a screen shot from the MIXER TAB on FaceBook it didn't look right anyway.

Using Newton's 1st and 3rd laws to rotation:
Torque = Inertia x Angular-Acceleration
An additional anticlockwise torque applied by the motors to the props will result in an additional clockwise torque applied to the quad. Which will provide a clockwise yaw.

Phil-MC, 2024-01-14:
@DzikuVx 
Pawel Spychalski. I see your name on the original doc. Any Comments,

Phil-MC, 2024-01-14:
This is the image posted by Oran Tao.
As he points out the +1 and -1 values in the table would not cause a clockwise YAW. "The motor number 1 and 4 should be '1' and number 2 and 3 should be '-1' "

[image]

DzikuVx, 2024-02-02:
There is no problem over here. Props out configuration is done either by switching Props-out switch or modifying the mixer, but not both at once.
Enabling `props-out` switch does not modify the mixer. Modifying a mixer does not switch `props-out` switch. Those are independent settings

closed 2024-02-02 by DzikuVx