iNavFlight/inav #9653: Autolaunch topic with 7.0.0 compared to 6.1.1 (fixed Wing, Funjet)

druckgott, 2024-01-20:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior
With 6.1.1 after some optimisations the autolaunch worked quite good with the fixed wing funjet
Now I updated to 7.0.0 and the funjet rotates quite much. It´s not realy smoth and i nearly crashed.

## Steps to Reproduce
Flash 7.0.0 with same diff of 6.1.1

## Expected behavior
same behaviour like 6.1.1

## Suggested solution(s)
attached the logs it looks so different see below
[LOG00045.TXT](https://github.com/iNavFlight/inav/files/13998649/LOG00045.TXT)

Is there anything changed between the autolaunch behavior on both verisons

## Additional context
7.0.0 diff
[INAV_7.0.0_cli_FUNJET_20240114_171807_diff.txt](https://github.com/iNavFlight/inav/files/13998639/INAV_7.0.0_cli_FUNJET_20240114_171807_diff.txt)
6.1.1 diff
[INAV_6.1.1_cli_FUNJET_20240114_170544_diff.txt](https://github.com/iNavFlight/inav/files/13998640/INAV_6.1.1_cli_FUNJET_20240114_170544_diff.txt)

- FC Board name and vendor:
- Speedybee Wing F405 fixed wing
- INAV version string:
- 7.0.0
[LOG00057.TXT](https://github.com/iNavFlight/inav/files/13998653/LOG00057.TXT)
- 6.1.1
[LOG00044.TXT](https://github.com/iNavFlight/inav/files/13998648/LOG00044.TXT)

MrD-RC, 2024-01-20:
Nothing has changed for autolaunch in 7.0.

druckgott, 2024-01-20:
Are there any other changed which can have influence the Autolaunch roll behavior

druckgott, 2024-01-21:
Here is some refacroting where Code in Autolaunch is changed or not

https://github.com/iNavFlight/inav/pull/9287/files

Jetrell, 2024-01-21:
>     * 7.0.0
>       [LOG00057.TXT](https://github.com/iNavFlight/inav/files/13998653/LOG00057.TXT)
> 
>     * 6.1.1
>       [LOG00044.TXT](https://github.com/iNavFlight/inav/files/13998648/LOG00044.TXT)

Looking at the logs.. It appears from the 7.0 log, that you moved the roll stick to full deflection, straight after `launch_in_progress` began... Which cancelled auto launch. And caused it to roll rather abruptly from looking at the `setpointRate[roll]` and rapid i-term roll accumulation.

[image]

druckgott, 2024-01-21:
the point is that on 7.0.0 it rolls much more then on 6.1.1 is the feeling. We did many tests on 6.1.1 to get it to a good behavior. Then i flashed to 7.0.0 and all 3 starts it rolls quite mutch and I put my handy to the stickts to roll it back.
[LOG00057.TXT](https://github.com/iNavFlight/inav/files/14001071/LOG00057.TXT)

Jetrell, 2024-01-21:
As soon as you move the stick past `nav_fw_launch_abort_deadband`. It will cancel auto launch. And in your case, become Horizon mode.
So we can't be seen from logs 57, what would have happened if auto launch had continued.
While in the log 44 from 6.1.1. Auto launch was completed.

> Here is some refacroting where Code in Autolaunch is changed or not
> 
> https://github.com/iNavFlight/inav/pull/9287/files

That just looks to be cleanup for multirotor throttle tilt compensation.

But the throttle scaling was refactored.. Including launch idle throttle changes. https://github.com/iNavFlight/inav/pull/9287
So it may be providing less available throttle in 7.0, with the same launch throttle setting from 6.1.1... Which might explain why the roll axis felt lazy at launch.

druckgott, 2024-01-21:
Ok why is this done does this makes sence or whata the background?

b14ckyy, 2024-03-20:
that change only affects multirotor not fixed wings.

closed 2024-03-20 by druckgott