iNavFlight/inav #9655: Better control over mannual alt control

Lalbabu2001, 2024-01-21:
## Current Behavior

when you limit the vertical speed of quad because you don't want to utilise full throttle for safety reasons using advance tuning tab the quad adopt an oscillating behaviour because it measures the vertical speed using baro sensor and it's not enough accurate there always some deviation it needs some time to measure the speed and says hey we have come more height let's go down and jump in case of sink
## Desired Behavior
quad should not took into account the speed in case if mannual throttle it just note

## Suggested Solution
the quad should not take account baro reading while mannual input and there should be some specific rates for throttle

## Who does this impact? Who is this for?

## Additional context

MrD-RC, 2024-01-21:
I believe this may help https://github.com/iNavFlight/inav/pull/9654

sensei-hacker, 2024-01-21:
While the fox Darren mentioned may reduce oscillations, I'm curious to make sure I understand what you're asking for.  It may be that what you want is already available.

It's unclear if you are wanting to reduce the throttle rate, or do something different with the climb limiter.

On the one hand, you said:
> there should be some specific rates for throttle

You can either back off the rate in your radio, or reduce the mix to less than 100% throttle, if you choose to.

You also said:

> when you limit the vertical speed of quad ...  using advance tuning tab

> quad should not took into account the speed in case if mannual throttle it just note
...
> the quad should not take account baro reading

That sounds like you're asking for INAV to enforce a maximum climb rate - without measuring the climb rate. Do I understand that correctly?  "Don't let it climb at more than 3 meters per second, and do that without looking at how fast it's climbing"?

b14ckyy, 2024-03-20:
Try 7.1 RC1. It has new sensor fusion that should provide more stable altitude control.

b14ckyy, 2024-03-22:
Closing for now. If you still have issues with 7.1, we can reopen.

closed 2024-03-22 by b14ckyy