iNavFlight/inav #9659: Terrain data for showing AGL

error414, 2024-01-22:
Hi, 

I play with rangefinders, but rangefinders are expensive, inaccurately, or both. 

Solution could be implement into inav terrain data saved on SD card similar to ardupilot. https://terrain.ardupilot.org . Precise for ardupilot is 30 or 10m, it's not enought for mountains, so would be nice to implement 1m precission from GeoTiff as well, lot of countries provide data for free. 

At start AGL can be showed only in OSD, no any automations. 

https://discuss.ardupilot.org/uploads/default/original/3X/5/e/5e1711b608419ec2c58a96824a0c281168ebe38d.png

b14ckyy, 2024-03-20:
Some day in the future, when we have a SD card driver that is capable of reading and writing blackbox, maybe. And yes I am all down for that :)

error414, 2024-05-20:
> Some day in the future, when we have a SD card driver that is capable of reading and writing blackbox, maybe. And yes I am all down for that :) Especially as it would come in handy for a 8.0 feature thats in the works.

I'm not sure what do you mean about SD card driver. 

Maybe it's naive idea but current SD driver (AsyncFatFS + sdcard_spi) supports. 

`afatfs_fopen
afatfs_fseek
afatfs_fread
`

file is represented by afatfsFilePtr_t so I would say it's possible to open multiple files. We don't need possibility to write file over mass storage to SD card. You can take SD card, put it to PC, copy "NXXeXXX.dat" file to root of SD card. 

Then load "mavlink grid" could be easy (2048 bytes).

b14ckyy, 2024-05-20:
@error414 I don't know the technical details but the current way INAV access the SD card has a lot of issues. 
We can't even write blackbox at 1000Hz or more reliably on most MCUs without getting data errors. Taking cycle times to read data while blackbox is writing, does not seem to work reliably at all. That's what a few devs have told me.

sensei-hacker, 2024-05-20:
> Maybe it's naive idea but current SD driver (AsyncFatFS + sdcard_spi) supports.

> afatfs_fopen afatfs_fseek afatfs_fread 

> a file is represented by afatfsFilePtr_t so I would say it's possible to open multiple files.

It's *possible*. Note that the SPI interface is much slower than the SD interface, which requires dedicated SD hardware. That may or may not be an issue. The SPI interface uses the same wire for commands as it does for data, so switching over to read, then reading data, switching back to write and writing some blackbox data will take time.

Keep in mind also F722 MCUs have a total of 512K flash and we've already had to remove features from the F722 builds in order to fit INAV in 512K. So a few more bytes matters - if the additional functions are another kilobyte that's an issue.

You are more than welcome to play with this and do a POC.