iNavFlight/inav #9662: Questions about NMEA protocol

do1234521, 2024-01-22:
As we know that in INAV7.0, the NMEA protocol is not supported anymore.  Only the UBLOX series protocol is supported and we have to use the UBLOX based gps module.
Here comes the question, what if we wanted to use  the gps module based on other chips?  
There are many options on the market, and the performance of them are good as well.  Some support L5 channel and some support RTK. And many of them is very cheap when compared to UBLOX modules.
Most of them can use 10Hz refresh rate and the worry about the refresh performance is not any problem.  But this modules mostly can only use NMEA protocol as output with high baud rates. 
So, will the development team consider to re-use the NMEA protocol in the future? Or is there any solutions to support more modules?

sensei-hacker, 2024-01-22:
> There are many options on the market, and the performance of them are good as well.

Any in particular you have in mind?

do1234521, 2024-01-23:
> > There are many options on the market, and the performance of them are good as well.
> 
> Any in particular you have in mind?

Yes, i am in China and there are lots of different gps modules on the market.
You can get a TAU1201 module by ALLYSTAR which support L5 at the price of about 10 USD dollars. A RTK module is about 20-30 dollars. 
As for cheap modules, you can buy an AT6558 based module which intergrated a FLASH inside at less than $2, and these modules can support 10Hz NMEA outpot.
Others include all kinds of 4G IoT based module which have gps functions, and the price is very cheap too, mostly is under $10
There are also some others avaliable but these modules are cheap and easy to get. You can go to Aliexpress and search for these.

majianjia, 2024-02-17:
well, i am looking at my 10x $2 old gps modules that only support nmea.