iNavFlight/inav #9668: Induced signals in other chanels when nav mode is used

terokorn, 2024-01-24:
## Current Behavior
I am working on a boat with differential thrust using some programming settings shown in a video by Pavel. So Servos 1 and 2 I am using for the engines. Servos 3 and 4 I use for doors located on the boat. The system includes Matek 411WTE, Matek M8Q-5883. Whenever a  Navigation mode is set (NAV_WP or NAV_RTH) the servo channels 3 and 4 randomly get a signal, and the servos start uncotrolled movement. I have triyed to keep them centered with a logic condition, and that have improved a bit this uncontrolled movement, but it is still there. You can see the logic conditions I used for that on the picture attached. In addition, I am attaching a diff all file (it is project in progress, so some stuff might look odd).
[image]
[INAV_Diff_All.txt](https://github.com/iNavFlight/inav/files/14034906/INAV_Diff_All.txt)
The video attached shows the issue with the logic condition for the servos removed ( so it is more easier to spot the issue)
Link to video as it is to big :( - https://drive.google.com/file/d/1gyH6IJlvqxuxzLTEelFRRCoIa8L_UZeu/view?usp=sharing

## Steps to Reproduce

1. Use manual mode to control the engines
2. No parasite signal 
3. Activate NAV WP or NAV_RTH
4. Observe Parasite signal to servos 

## Expected behavior
No noise signals are transferred to other channels.

## Suggested solution(s)
I have not found a way to work with differential thrust in the best way without programming logic. And more precisely, the system will not reverse one of the engines in a turn but will only stop it. This reduces the maneuverability of the boat or rover, and it will be good if when boat or rover is used, the engines to be operating this way.

## Additional context

[INAV_Dump.txt](https://github.com/iNavFlight/inav/files/14034974/INAV_Dump.txt)

---

# version
# INAV/MATEKF411TE 7.0.0 Dec  5 2023 / 10:36:06 (895a4f31)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor: Matek F411WTE
- INAV version string:

DzikuVx, 2024-02-02:
This is correct, it's not random. NAV engine works by overriding rc channels. And from your code I see that it's using roll and pitch as input for LC 20

closed 2024-02-02 by DzikuVx

terokorn, 2024-02-02:
Hi Pavel, Thanks for the comment. Yes, I am using roll and pitch channels, but for servos not connected to the boat's directional movement. So I wonder if there is a way to avoid this. LC20 is a desperate attempt of mine to stop that movement when the pilot does not command it :). Can you share where I am mistaken?