iNavFlight/inav #9674: Automatically estimate 'Hover throttle' value for Multirotors to ensure that weight changes do not adversly affect performance of POS HOLD etc.

Painless360, 2024-01-25:
At the moment, hover throttle needs to be set as a discreet value in configurator (1250 default). This needs to be fine tuned by the pilot using a test hover and then manually edit the value. However 'hover throttle' changes during the flight as the battery charge level changes and adding an action camera or different weight of battery will affect this value and cause the MR to rise/drop when modes like POS HOLD is used. This means that the pilot needs to change this value when they add a GoPro, or take it off, or add another addition that affects the weight and thrust of the model.

In Ardupilot, ArduCopter monitors the throttle and movement of the MR and sets this value dynamically allowing for safe transitions to navigation flight modes with very smooth transitions. It would be great if INAV could do something similar.

This would impact all pilots using INAV on a MR and trying out the navigation modes. Changes in model weight, props and battery voltage/charge wouldn't affect the smooth transition of the model.

Thank you for considering this. Happy flying!

JulioCesarMatias, 2024-01-29:
I have tried this in the past >> https://github.com/iNavFlight/inav/pull/8507

VitaliiShevchenko, 2024-02-02:
By the way @Painless360 , in the configuration in Other features tab have radio button for switch on "Throttle voltage compensation". Did you tried switch on this radio button?

JulioCesarMatias, 2024-02-02:
> By the way @Painless360 , in the configuration in Other features tab have radio button for switch on "Throttle voltage compensation". Did you tried switch on this radio button?

This feature works and delivers a different result than what was described in ISSUE. Therefore, unfortunately this feature does not help in this case.

rts18, 2024-04-02:
> I have tried this in the past >> #8507

I tried that commit. Unfortunately it did not work.
After assessing the way iNAV handles vertical position. I see why `nav_mc_hover_thr` is a required setting. 
Its a baseline, so the estimator can account for sensor inaccuracy.
Barometer, accelerometer and GPS vertical position can all drift.
Only with the use of a lidar or radar sensor could such a feature work with consistency, without a total overhaul of the vertical position estimator.

JulioCesarMatias, 2024-04-05:
> I tried that commit. Unfortunately it did not work. After assessing the way iNAV handles vertical position. I see why `nav_mc_hover_thr` is a required setting. Its a baseline, so the estimator can [...]

You might want to try this again, changing the "hover_throttle" variable from uint16_t to float.