iNavFlight/inav #9680: Add custom motor sequence

Zhu-hongbing, 2024-01-30:
In the current version, 7.0 only supports modifying the motor sequence for x4. We hope that x8 and y6 can also be changed in the future

sensei-hacker, 2024-01-30:
Note you _can_ do as many motors as you want in any positions you want. It just doesn't have a pretty interface to do so with a couple clicks.

Suppose you want motor 6 to do what motor 1 is doing. You want to swap the behaviors of motor 1 and motor 6. In the motor mixer, look at the pitch, roll, and yaw values for motor 1 and jot them down. 

 Copy the pitch, roll, and yaw values from motor 6 to motor 1.  Then replace the values on motor 6 with the ones that were on motor 1.

sensei-hacker, 2024-01-30:
When setting up your custom motor positions:

Motors in the rear get positive pitch values. Motors in the rear, negative pitch values.

Motors on the left get positive roll. Motors on the right get negative roll.

closed 2024-03-22 by b14ckyy