iNavFlight/inav #9685: CRSF/ELRS telemetry issue on Matek F405-wing V2

krugs44364, 2024-02-01:
## Current Behavior
I recently purchased a Matek F405-wing V2 and I am unable to get CRSF/ELRS telemetry working. The RX channels works. The only telemetry back to the radio (TS16S) if the receiver telemetry. Nothing from the FC.

## Steps to Reproduce

1. Flashed Inav 7.0
2. Connect Radiomaster Rp# ELRS receiver on UART2
3. set parameters as per Matek's recommendations product data sheet for the CRSF protocol
4.

## Expected behavior
Full FC telemetry back to TX16S radio

## Suggested solution(s)
I suspect the firmware has a bug that is not sending Telemetry data to the TX2 pin for the CRSF protocol.

## Additional context
I have connected the same Rp3 rx to 2 other Matek FCs I have on hand, an F405-WMN and an F465-wing, both flashed with version 7 and the telemetry works. Used the same model on the TX16s as well, which proves that the ELRS setup is correct.
I have also tried another F405-wing V2 and it has the same behaviour. So that rules out a faulty FC. To further prove that the FC hardware is ok, I also set up a Radiomast R161 receiver using SBUS and Smartport telemetry on UART2, and the telemetry works.
As a final test and out of desperation, I then flashed Arduplane firmware on one of the F405-wing V2s and got working telemetry with the ELRS receiver.
I also tried version 6.0 and the issue was the same.

Some background: I initially purchased one Matek F405-wing V2 from AliExpres in December 2023. When I couldn't get that to work I purchased a 2nd one from PhaserFPV in Sydney to rule out a faulty FC.
 

[INAV_7.0.0_cli_20240202_063952.txt](https://github.com/iNavFlight/inav/files/14132082/INAV_7.0.0_cli_20240202_063952.txt)

---

# version
# INAV/MATEKF405SE 7.0.0 Dec  5 2023 / 10:29:45 (895a4f31)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor: Matek F405-wing V2
- INAV version string: Version 7

sensei-hacker, 2024-02-02:
From the Matek product page:

```
Softserial_tx1 is enabled on Tx2 pad by default for Frsky SmartPort telemetry.   If using CRSF protocol receiver(TBS, ELRS etc),  CPU based serial port(softserial) should be disable in INAV configurator.
```

Did you turn off soft serial?

If so, lack of telemetry from the FC to the reciever with ELRS generally indicates a fault in the connection between FC tx pad and the receiver rx pad. Often, this is caused by a wire being broken within the insulation. This can be caused when the wire was previously caught under a screw / standoff, or if it gets kinked.

It can also be a soldering issue, but in your case I suspect a bad wire is more likely, if you turned off SoftSerial.

krugs44364, 2024-02-02:
Hi Sensei.
Yes, I have.
Last night I also succesfully tested the board with a Radiomaster R161
receiver which uses SBUS in the board's SBUS pin and Smartport for
telemetry which uses the TX2 pin albeit with softserial.
However, it proves the board's hardware works.
To prove the ELRS receiver, I connected the same receiver to another FC
(Matek F405-WMN) I have and the telemetry works.
That leaves only firmware in my opinion. I have tried to flash Inav 6 but
the problem was the same.
Lastly I flashed Arduplane to the F405-wing VE board and the telemetry via
the ELRS receiver worked. This again proves to me it is the INAV firmware
that is the issue.

sensei-hacker, 2024-02-02:
I would triple check with SoftSerial definitely turned off.

The firmware doesn't have specific code for "CRSF on uart2". Uart2 could be broken in the firmware, or CRSF could be broken, but "CRSF on uart2" doesn't exist in the firmware, and therefore could not be broken.

In your shoes, after triple checking that soft serial really is off, I would test it on uart 5 or uart 6 with a new piece of wire.  A wire broken inside the insulation (or plug is the most common cause of this issue, on all boards.  Obviously on that particular board, not realizing you need to actively turn off SoftSerial is a bigger issue.

krugs44364, 2024-02-02:
Hi Sensei,
Thank you for your response.

Yes, I have also been talking to support from my local supplier, and from
that, I have checked that soft serial is off multiple times. I have also
tried  UARTS 5 and 6.

If you read my previous response again you will see that I have managed to
establish telemetry with the same rX and "wire" when I flashed Ardupilot
firmware on the FC. I think that rules out any possible "wire" issues.

You are saying CRSF on UART2 doesn't exist but I'm sure you can understand
that what I mean is I have the receiver connected to UART 2 and have
selected CRSF protocol.

The other point I mentioned that you may have missed is that I used the
same receiver and hence the same "wire" on another FC that I have (a Matek
F405-WMN) and the telemetry works. So definitely not a wire issue.
Furthermore, I also tried with another receiver (betafpv nano 900Mhz) but
was unable to get telemetry going on the Matek F405-wing V2 while it works
on the F405-WMN.

So the point is I have proven beyond any doubt that is it not hardware, not
a wire issue, and not a receiver issue. The  F405-Wing V2 and the F405-WMN
use different firmware loads. So you are correct it is most likely that
CRSF tlemetry is broken in the firmware. Since it gives the same issue on
UART 5 and 6 is is unlikely that it is a problem with UART2.

As another, point., I also tried the same wire and receiver on an old Matek
F465-wing I have and again the telemetry works!

NB. I have got a lot of experience with setting up ELRS on Matek FC's and
have never run into a problem that didn't have a simple solution except now.

I appreciate your assistance and I understand you want to make sure it is
not something silly that I have missed but I have provided enough info here
that should make it clear that I haven't missed anything.

I've attached some screen shots:

[image: Receiver.JPG]
[image: ports.JPG]
[image: Config.JPG]
Thanks for your help so far.

sensei-hacker, 2024-02-02:
Email replies to Github issues don't carry the screenshots. Can you please attach those via the Github web site.
Particularly your ports tab, because the attached diff indicates there may be something interesting there, but I would like to confirm visually.  Also a dump could be handy.

> If you read my previous response again you will see that I have managed to
establish telemetry with the same rX and "wire" 

Previously, you didn't mention which side you desoldered the wire from - the FC or the receiver.
Moving a receiver from one FC to another, one could leave the old wire on the FC, or on the receiver.

> So the point is I have proven beyond any doubt that is it not hardware, not
a wire issue, and not a receiver issue. The  F405-Wing V2 and the F405-WMN
use different firmware loads. So you are correct it is most likely that
CRSF tlemetry is broken in the firmware. Since it gives the same issue on
UART 5 and 6 is is unlikely that it is a problem with UART2.

Assuming all of your tests are correct, it would mean that:
It's not a problem with that UART.
It's not a problem with the hardware
It's not a problem with the receiver
It's not a problem with the wiring

Therefore, either:
A. It's just that crsf doesn't work in INAV.  Meaning ELRS doesn't work with INAV.  
B. You've "proven" something that isn't true. (testing error)
C. Configuration problem

A is obviously not the case.  Thousands of people use it every day.

However, your diff indicates C is worth looking at a bit closer. In particular the ports config.

sensei-hacker, 2024-02-02:
Here's something you can try, BTW:

Turn SoftSerial back on. Save and reboot.
I think you will see smartport telemetry will be enabled on Softserial by default.
Remove smartport from softserial. Save and reboot.
Disable softserial.

Discover new telemetry sensors.

It's possible that the default config is trying to send telemetry to a port that does not exist.

krugs44364, 2024-02-02:
Hi Sensei,
I don’t de-solder and re-solder when I move the receiver from one FC to
another. I is on a DuPont 4 pin connector.

I will send a dump.
Should I also send a dump of the F405-WMN where the telemetry works for
comparison?

Kind regards,
Carel Kruger

sensei-hacker, 2024-02-02:
Ah that makes sense, Carel. I fly mostly fixed wing, but don't use Dupont because they are so darn big.
It didn't occur to me you were likely using Dupont.

A dump of the WMN for comparison sounds good too.

krugs44364, 2024-02-02:
Thanks, I will give it a go.

Kind regards,
Carel Kruger

krugs44364, 2024-02-02:
Dump files:

-- 
Kind regards,
Carel Kruger

dump

# version
# INAV/MATEKF405SE 7.0.0 Dec  5 2023 / 10:29:45 (895a4f31)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# features
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -RPM_FILTERS
[... 2594 lines cut ...]
osd_layout 3 28 23 11 H
osd_layout 3 29 23 12 H
osd_layout 3 30 1 13 H
osd_layout 3 31 0 10 H
osd_layout 3 32 12 1 H
osd_layout 3 33 6 2 H
osd_layout 3 34 18 2 H
osd_layout 3 35 1 5 H
osd_layout 3 36 1 5 H
osd_layout 3 37 1 6 H
osd_layout 3 38 1 7 H
osd_layout 3 39 1 5 H
osd_layout 3 40 1 2 H
osd_layout 3 41 1 8 H
osd_layout 3 42 1 7 H
osd_layout 3 43 0 0 H
osd_layout 3 44 0 0 H
osd_layout 3 45 0 0 H
osd_layout 3 46 3 6 H
osd_layout 3 47 3 7 H
osd_layout 3 48 23 7 H
osd_layout 3 49 23 6 H
osd_layout 3 50 0 0 H
osd_layout 3 51 12 2 H
osd_lay

krugs44364, 2024-02-02:
No luck.

kd8b, 2024-02-04:
All Telemetry working correctly for me. Matek-F405-WING-V2 -Matek M10q-5883, GPS-Compass, BetaFPV, SuperD RX2400 ELRS.
     I just set up a Matek f405 wing v2 on a vtail plane with 7.0.0. I too saw the issue with no elrs telemetry from the controller. I did find in my case, that the soft serial was in fact set to smartport by default and disabled it. Problem corrected. 24 items of telemetry after re-running discover new. Telemetry Output enabled in config. ELRS running 12 ch mixed mode. Crossfire enabled in yaapu config in my case as I run INAV Telemetry widget as well. RX on UART2 GPS on UART3. It can work,

krugs44364, 2024-02-04:
Hi kd8b,
Thanks for that. Would you be prepared to share your INAV dump file with me?

krugs44364, 2024-02-04:
Hi kd8b,
That is good to hear (i think).
Would you share your INAV dump file with me?

kd8b, 2024-02-04:
[Dump - Talon GT Rebel-Matek F405 Wing - v2 - INAV_7.0.0_cli_20240204_183924.txt.zip](https://github.com/iNavFlight/inav/files/14158994/Dump.-.Talon.GT.Rebel-Matek.F405.Wing.-.v2.-.INAV_7.0.0_cli_20240204_183924.txt.zip)
I hope this is the correct way to do this as I have never uploaded to Github before.

krugs44364, 2024-02-04:
Thanks got it.

kd8b, 2024-02-04:
My INAV build is untested in flight so consider this a test bench setup until verified. New to INAV Airplane but have multiple INAV quad builds

krugs44364, 2024-02-04:
Thanks, at this point mine is also just test bench until I can get the
telemetry to work.

krugs44364, 2024-02-06:
Well, that didn't work either. For now, I will just have to run with Ardupilot on the F405-wingV2.

MrD-RC, 2024-02-06:
If you are getting control but no telemetry, it’s 1 of 2 things usually. 

1. The RX and TX are the wrong way around. 
2. The receiver configuration is wrong. Check in the Crossfire LUA to make sure those pads are set to CRSF TX and CRSF RX.

krugs44364, 2024-02-06:
I've ruled those out already. My receiver is on a duPont 4 pin connector, so I just move it to a F405-WMN and the telemetry works.
So the radio, Lua, receiver, wiring etc. is common only change is the FC and obviously the firmware. 
Using the same hardware, when I flash Arduplane on to the F405-wingV2 the telemstry also works albeit with Yaapu on my TX16s instead of the Inav Lua.

MrD-RC, 2024-02-06:
Then it must be a setting on the receiver page. CRSF works just fine. It can only be setup if you’ve proven all the hardware.

krugs44364, 2024-02-07:
Hi Darren,
Yes, I agree it just works - normally. I’ve done lots of setups without
issues.

This is what my receiver tab looks like.
[image: image.png]
I've also tried changing the Serial receiver half duplex setting to off and
still nothing.
The way this is set is identical to how it is on my F405-WMN where
telemetry works.

This has got me stumped and quite frustrated.

Kind regards,
Carel Kruger

krugs44364, 2024-02-20:
I finally resolved the issue by updating the TX module firmware to the latest available. Still a bit of a puzzle why the older firmware would work on some older FC's but not on a new Matek F405-wingV2.

b14ckyy, 2024-03-20:
@krugs44364 can you please do me a favor and check if it still works with INAV 7.1 RC1 or this inifficial RC2 I made? [https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl](https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl&fbclid=IwAR1Xl_HEmQKRJMXpcgoZrq-1eS4M6BPGAywUds-Ca4UUZYCdqI2Gkm7kr2E)

do a fresh telemetry sensor scan and check if you see a Vspd sensor in that list.

krugs44364, 2024-03-20:
Ok, I'll check.

krugs44364, 2024-03-21:
Hi,
It work on both 7.1RC1 and RC2 and I can see Vspd.

b14ckyy, 2024-03-21:
awesome thanks. I will close here then.

closed 2024-03-21 by b14ckyy

Raywnc, 2025-07-23:
I was able to resolve my telemetry issue by reading this post.

Thank you all, Ray