iNavFlight/inav #9692: Target Question: AtomRC F405 NAVI Mini

HazzaHFPV, 2024-02-04:
Concerning the AtomRC F405 NAVI Target.

AtomRC have got two FCs: The F405 NAVI and the F405 NAVI **Mini**. They both use the same target, however the MINI does not have an SD-Card slot, but an onboard flash of 16MB. As the target is configured for the SD-Card slot, Blackbox cannot be used on the F405 NAVI Mini. I suspect a new target needs to be added for the mini?

I also see there is a new target "AtomRC F405 NAVI delux" but I cant find that board anywhere on the net?

DzikuVx, 2024-02-04:
Have you tried to contact AtomRC customer support?

HazzaHFPV, 2024-02-04:
I have written them a message, but I think we are moving into the time of chinese new year and therefore I don't expect an answer soon.

What I also found was that when I select "Serial" as blackbox, then "save and reboot", as soon as I try to go back to the Blackbox-Tab INAV becomes inresponsive and the FC has to be rebooted by removing power and USB to access it again.

I was hoping to just use an external logger that I have laying around and wanted to solder that to UART3.

HazzaHFPV, 2024-02-13:
After trying for a long time, I seem to have managed to compile a custom target for the AtomRC F405 NAVI MINI, with working Blackbox. I will attach the hex-file here. If you choose to use it, please test with caution, as I cannot guarantee that there aren't any bugs or other things that might not work, even if it seems fine on my unit.
[INAV_7.0.0_ATOMRCF405NAVI_MINI.zip](https://github.com/iNavFlight/inav/files/14258964/INAV_7.0.0_ATOMRCF405NAVI_MINI.zip)

mmosca, 2024-02-23:
You may want to share what modifications you had to do, in case others are looking for it in the future.

closed 2024-03-22 by b14ckyy

jankiefer, 2024-11-05:
> After trying for a long time, I seem to have managed to compile a custom target for the AtomRC F405 NAVI MINI, with working Blackbox. I will attach the hex-file here. If you choose to use it, please test with caution, as I cannot guarantee that there aren't any bugs or other things that might not work, even if it seems fine on my unit. [INAV_7.0.0_ATOMRCF405NAVI_MINI.zip](https://github.com/iNavFlight/inav/files/14258964/INAV_7.0.0_ATOMRCF405NAVI_MINI.zip)
> 
> Edit: and up to this day, I have not received an answer from AtomRC addressing my issue. If you are in the market for a small INAV Wing Flight Controller, the new Speedybee Wing Mini would be the way to go.

Is tehre any updated firmware available for 7.1.2? I can't use the 7.0.0 as configurator asks me to update firmware...

b14ckyy, 2024-11-05:
No. Configurator asks you to update configurator. You can disable that check in the Configurator settings (top right). 
As stated above, you should contact AtomRC regarding a new target.

jankiefer, 2024-12-09:
@HazzaHFPV Could you share how you did configure the target to use the memory flash chip? I managed to modify and compile other targets, so this one would be the next on the list...

theramman, 2024-12-31:
I was able to get this to work in Ardupilot, I needed to change the SPI pin for the SDcard to flash and in ardupilot enable dataflash logging. I hope this helps.

Noramas, 2025-01-05:
Has anyone managed to enable the blackbox feature on the F405 NAVI Mini? What configuration changes are required before compilation?

kijan007, 2025-03-26:
@Noramas , @mmosca 
- compiled firmware with working flash, Code is in fork for now - https://github.com/kijan007/inav/tree/kijan007-ATOMRC405MINI

Jake-JB, 2026-08-29:
Hi all, I'm not an experienced github user but I'm having this issue now where I want to use my blackbox on my AtomRC F405 Mini, not having the best luck trying to figure out how to use the code linked on this forum, could I get some help? Does this flash still work and how do I go about getting it working?