iNavFlight/inav #9699: Autolaunch behavior of elevator Ruder during throttle rise

druckgott, 2024-02-05:
## Current Behavior
If you have switched on autolaunch the elevator is moving down. 
Then you rais the throttle to idle speed and parallel the elevator is moving upwards (slow)

If you now throw the wing to early it get crashed (I did this several times because I did not realise the ruders are moving during throttle moving)

If you now alrady throw because motor is finished the aircraft is going deeply down to the earth. We could not reproduce our autolaunch with the wing until we find out this. Some times it went fast to the earth sometimes it starts very well. If we always wait now until the ruder are finished all starts are working well.

I did a Video. Why is the ruder moving up during movement of throttle up.

https://github.com/iNavFlight/inav/assets/18383738/b3aaa46a-abf1-4c2f-895f-7c0b5b6b80fd

## Steps to Reproduce
1. Start Autolaunch
2. move Up throttle from low to idle
3. Elevator is moving upwards very slow maybe idle is already reached and moving of elevator is not finshed.

## Expected behavior
1. Start Autolaunch
2. move Up throttle from low to idle
3. Elevator is moving upwards very slow (don´t know why) 
4. Motor starts at idle if Elevator is finished

## Suggested solution(s)
3. Elevator is moving upwards very slow (don´t know why) 
4. Motor starts at idle if Elevator is finished

## Additional context

- FC Board name and vendor: Speedybee Wing 405 V3
- INAV version string:

```
diff

# version
# INAV/SPEEDYBEEF405WING 6.1.1 Jun 13 2023 / 12:28:36 (42ced253)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 80 0 -1
smix 1 1 1 85 0 -1
smix 2 2 0 -90 0 -1
smix 3 2 1 95 0 -1
smix 4 3 6 100 0 -1

# Outputs [servo]
servo 1 1000 2000 1468 100
servo 2 1000 2000 1546 -100
servo 3 1037 1980 1665 -100

# safehome

# features
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map
map TAER

# Ports
serial 3 1 115200 115200 0 115200
serial 4 33554432 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 2 900 1500
aux 1 1 1 900 1300
aux 2 2 1 1300 1700
aux 3 12 0 1300 1700
aux 4 11 0 900 1200
aux 5 10 3 900 1200
aux 6 3 3 900 1200
aux 7 13 4 1650 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 46 0 H
osd_layout 0 1 12 0 H
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 7 2 19 V
osd_layout 0 9 1 2 H
osd_layout 0 11 40 1 V
osd_layout 0 12 13 19 V
osd_layout 0 13 39 9 H
osd_layout 0 14 2 18 V
osd_layout 0 15 21 19 V
osd_layout 0 20 6 1 V
osd_layout 0 21 6 0 V
osd_layout 0 22 26 7 V
osd_layout 0 23 29 19 V
osd_layout 0 25 51 8 H
osd_layout 0 26 46 0 H
osd_layout 0 28 28 18 V
osd_layout 0 30 13 15 V
osd_layout 0 31 7 18 V
osd_layout 0 32 39 0 V
osd_layout 0 85 39 9 V
osd_layout 0 86 1 1 H
osd_layout 0 87 6 2 V
osd_layout 0 109 20 18 V
osd_layout 0 110 22 18 H
osd_layout 0 111 21 18 H
osd_layout 0 112 13 18 V

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_y = -17
set gyro_zero_z = 2
set ins_gravity_cmss =  980.428
set acc_hardware = ICM42605
set acczero_x = -3
set acczero_y = -8
set acczero_z = 22
set accgain_x = 4100
set accgain_y = 4103
set accgain_z = 4094
set align_mag = CW0FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set serialrx_halfduplex = OFF
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set align_board_pitch = -4
set platform_type = AIRPLANE
set model_preview_type = 8
set small_angle = 180
set ahrs_inertia_comp_method = ADAPTIVE
set applied_defaults = 4
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set deadband = 1
set yaw_deadband = 1
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_wp_max_safe_distance = 500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 7000
set nav_rth_home_altitude = 3000
set nav_fw_bank_angle = 55
set nav_fw_loiter_radius = 3000
set fw_loiter_direction = YAW
set nav_fw_launch_max_angle = 180
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_spinup_time = 5
set nav_fw_launch_timeout = 8000
set nav_fw_launch_max_altitude = 8000
set nav_fw_launch_climb_angle = 35
set osd_video_system = BFHDCOMPAT
set osd_camera_fov_h = 122
set osd_camera_fov_v = 93
set osd_hud_radar_disp = 3
set name = FUNJET
set pilot_name = MW
set tz_offset = 60
set tz_automatic_dst = EU

# profile
profile 1

set fw_p_pitch = 20
set fw_i_pitch = 5
set fw_d_pitch = 4
set fw_ff_pitch = 110
set fw_p_roll = 24
set fw_i_roll = 3
set fw_d_roll = 7
set fw_ff_roll = 45
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 550
set max_angle_inclination_pit = 550
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.300
set nav_fw_pos_z_p = 35
set nav_fw_pos_xy_p = 85
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 25
set rc_yaw_expo = 30
set roll_rate = 40
set pitch_rate = 10
set yaw_rate = 3

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 255
set vbat_warning_cell_voltage = 270
set battery_capacity = 4500
set battery_capacity_warning = 1350
set battery_capacity_critical = 900
set throttle_idle =  5.000
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1200

# save configuration
save

# end the command batch
batch end

# 

```

druckgott, 2024-02-05:
We have this issue because another Person is throwing the Modelle and we did Not realize this and thro it right after Motor was speed up 🙈

flavioyarade, 2024-03-17:
Hi guys. I have the same problem and I can't find the solution. Of the 4 throws, 1 goes straight to the ground, leaning a little to the left... I don't understand why he does this. Could you help me. Thank you so much!!

sensei-hacker, 2024-03-17:
> Hi guys. I have the same problem and I can't find the solution. Of the 4 throws, 1 goes straight to the ground, leaning a little to the left... I don't understand why he does this. Could you help me. Thank you so much!!

If you have the *same issue*, you can use the same solution - throttle up to get launch idle before you throw, rather than while throwing.

A more common launch problem is people kinda drop the plane, a weak little toss, rather than THROWING it. The throw is what makes the plane fly. The motor *keeps* it flying. So throw it like you want it to fly, not like you're tossing on the ground in front of you.

flavioyarade, 2024-03-17:
I have done exactly that. I throw it hard straight so that it detects the throw as soon as it leaves my hand. I do the acceleration 2 seconds before launching it. It has an idle of 1600, to ensure that it has good air flow and 2000 once it comes out of idle. I think I have done everything that is appropriate. Even so, 1 out of every 4 pitches goes to the ground... I can give you the DUMP as I have it programmed. It's a Dolphin. Thank you for your time and the prompt response, it is very helpful to me.

dump

# version
# INAV/SPEEDYBEEF405WING 6.1.1 Jun 13 2023 / 12:28:36 (42ced253)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 50 0 -1
smix 1 1 1 50 0 -1
smix 2 2 0 -50 0 -1
smix 3 2 1 50 0 -1

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1504 -100
servo 2 1000 2000 1492 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# features
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -RPM_FILTERS
feature -TELEMETRY
feature -CURRENT_METER
feature -REVERSIBLE_MOTORS
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
[... 1608 lines cut ...]
set turtle_mode_power_factor = 55
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1500
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 2000
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 2000
set nav_fw_launch_idle_thr = 1550
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

#

druckgott, 2024-03-17:
I now do it like this I will later Post my diff. It was recommended by some body in rcgroups.com and it works perfekt i have no crash anymore. I did so many Tests and repaied it more then 10 Times. 
https://github.com/iNavFlight/inav/assets/18383738/0a01583e-e3a1-465e-b02b-c1e1631cf392

druckgott, 2024-03-17:
Here are the two diffs for launch like video above. The Ruders on the start are moved down quite a much but do have no "grip" on the beginning so the funjet is moving upwards first.

[INAV_7.0.0_cli_FUNJET_20240303_095527_diff.txt](https://github.com/iNavFlight/inav/files/14628055/INAV_7.0.0_cli_FUNJET_20240303_095527_diff.txt)
[INAV_6.1.1_cli_FUNJET_20240302_112723_diff.txt](https://github.com/iNavFlight/inav/files/14628056/INAV_6.1.1_cli_FUNJET_20240302_112723_diff.txt)

flavioyarade, 2024-03-17:
Thanks for your data. I can't find what it could be. I don't think I have anything wrong with my configuration... I have other models that work perfectly with my same configuration... It could be some BUG in the firmware... I don't know.

b14ckyy, 2024-03-22:
To conclude: The control surface movement is normal when throttle stick is lifted for launch (I-Term kicks in)
If the plane rolls on launch its one of the following issues: 

- Control surfaces massively out of trim
- Torque roll by too much launch power
- unbalanced wings (from wing mounted equipment
- bad configuration or mixer

Will close the ticket at this point as all the options to check are mentioned.

closed 2024-03-22 by b14ckyy