iNavFlight/inav #9706: Regarding altitude hold issues in iNav

yxc407, 2024-02-09:
I apologize for my ignorance. I would like to inquire whether iNav's Althold is capable of altitude holding and stability solely relying on barometer like PX4 or APM. I have conducted several experiments with a quadcopter:

1.Solely relying on the barometer and enabling Althold mode independently. In this case, while altitude holding functioned properly, the control felt like manual mode.

2.Solely relying on the barometer and simultaneously enabling Althold and Angle modes. In this case, I have come close to achieving the expected results. However, I am unsure if it's an issue with the barometer. When I make minor adjustments to the throttle stick, I can achieve the desired effect. But when I make significant throttle adjustments, the quadcopter continues to ascend, and I cannot perform any throttle stick operations until a few seconds later. After checking sensor data, I suspect this might be a problem with the barometer's delayed response. I would appreciate confirmation on this matter, whether it's related to the sensor or if I should adjust the algorithm.

In addition to the barometer, I have also added a rangefinder and used Althold + Angle mode. The results were similar. However, if I only use the rangefinder, the Althold mode does not appear.

Any insights or suggestions would be greatly appreciated.

JulioCesarMatias, 2024-02-09:
Do you have a Discord account so we can chat?

sensei-hacker, 2024-02-09:
>  When I make minor adjustments to the throttle stick, I can achieve the desired effect. But when I make significant throttle adjustments, the quadcopter continues to ascend

The baro isn't fast, that's true.  ALSO make sure there is enough open cell foam over the baro.

The baro is heavily filtered for short-term noise, which makes the response slow.  *One* reason it has a lot of short term noise is because people fly with the props blowing right into the baro. That same airflow from the props can also cause it to continue to rise, as the props create higher pressure below to raise the aircraft. Higher pressure is read as lower altitude.

yxc407, 2024-02-13:
> The baro isn't fast, that's true. ALSO make sure there is enough open cell foam over the baro.
>
> The baro is heavily filtered for short-term noise, which makes the response slow. _One_ reason it has [...]

Sure, I'll give it a try. Thanks for responding.

yxc407, 2024-02-13:
> Do you have a Discord account so we can chat?
Sorry, I'm not very good at speaking English. Thank you for your kindness.

closed 2024-02-13 by yxc407