iNavFlight/inav #9707: No GPS Fix with BEITIAN BN-180/220 on Inav 7.0.1

blooddog70, 2024-02-09:
Hi, I have a problem with my Beitian BN-180/220 with INAV 7.0.1! I'm currently building my first wing, I already have several copters with BF running with the above-mentioned GPS. But I can't get any satellites with INAV, everything is connected correctly, the GPS symbol lights up blue, GPS is activated, the correct UART is assigned to the Ports tab, and I've already tried all of the baud rates. Protocol is UBLOX selected, Ground Assistance Type European EGNOS, GPS use GALILEO SATTELITES (EU), TimeZone Offset +01:00, AUTOMATIC DAYLIGHT SAVING MODE is EU. It also counts up the TOTAL MESSAGES, UPDATE TIME 5.00 Hz mostly, No ERRORS or TIMEOUTS. I have no idea what else I should try to make it work. Does anyone have an idea? Thanks in advance for all the helpful answers.

Status say that in CLI:

[image]

MrD-RC, 2024-02-09:
@sensei-hacker has basically summed everything up. Once the almanac downloads, things will speed up. If you don’t get any satellites outside after 20 minutes. You may want to try a different module. Maybe the antenna is faulty or there is a problem with the module. 

The communication between INAV and the GPS module is fine. I would recommend using UBLOX7 rather than UBLOX.

closed 2024-02-09 by MrD-RC

blooddog70, 2024-02-09:
ok and which baudrate should I take in the Ports Tab? 9600? OK i will test UBLOX7, why is it better to UBLOX Protocol?
I tested 3 different Modules, VTX antenna is far away....

blooddog70, 2024-02-09:
Should i do that:

https://30081112.s21i.faiusr.com/61/ABUIABA9GAAg0oPEpAYotI7PdA.pdf

MrD-RC, 2024-02-09:
115200

dekakrens, 2025-02-17:
I have the same problem. I use Flywoo Goku GM10 nano v3 and MicoAir M10G, both of them hard to get GPS lock with INAV 7.1. Also, I tried to update to INAV 8 and the problem is still there. Even if I get locks, and then I fly the GPS lock is gone. I know my GPS module is fine because I also tested it on other quads running betaflight and never had this problem. I also mount the GPS far from other interferences like VTX antenna, the wire is also shielded and grounded to provide more signal integrity. Is there anything else I can try? I really want to fly INAV and have plan to build fixed wing with INAV

sensei-hacker, 2025-02-17:
> I have the same problem. I use Flywoo Goku GM10 nano v3 and MicoAir M10G, both of them hard to get GPS lock with INAV 7.1.

You are testing outdoors, right? After downloading the almanac and emperides for at least 12.5 minutes? The M10 doesn't have flash to save things like the M8 and M9 do, so it can be slower on warm starts.

Please show us the configuration in the GPS tab and how you have things mounted.