iNavFlight/inav #9720: CRSF SNR OSD Element doesn't work - Analog OSD

malinhiles, 2024-02-11:
## Current Behavior
CRSF SNR OSD element can be enabled in the configurator, but it doesn't appear in the video feed. Both the LQ and RSSI elements work, but the SNR element does not work.

[image]
[image]

## Expected behavior
OSD element should work

 version
 INAV/MATEKF405SE 6.0.0 Mar 22 2023 / 10:18:18 (084101ac)
 GCC-10.2.1 20201103 (release)

OptimusTi, 2024-02-12:
It works as designed. You don't see the values unless they are within range. See the ? for description.

OptimusTi, 2024-02-14:
This can be closed, as it's not an issue and the feature works as designed.

closed 2024-02-27 by mmosca

OptimusTi, 2024-02-27:
I've been planning to document this better, but I'm having issues with the 8.0 configurator master.

malinhiles, 2024-02-27:
> It is working as expected. You don't see the values unless they are within range. Range is between -20dB and +4dB. If values are outside of that range, you don't need to see it. The upper limit can be set with osd_snr_alarm in the CLI. Default is +4dB.

Ahh i see. So if I wanted to see it all the time, i'd need to increase the upper limit of ```osd_snr_alarm``` so that the element's value is always below that threshold? But by doing this, does it also mean that the element will always be flashing on my screen because its value is now below my increased alarm value for ```osd_snr_alarm```?

>If values are outside of that range, you don't need to see it.
Now that I know, Im curious as to why this is the default behaviour. I don't know why I would only want to see the SNR when it gets low, similar to why I wouldn't want to only see my flight pack battery voltage if it gets below a threshold. Even regular RSSI acts like a regular OSD element, only RSNR has this behavior.

OptimusTi, 2024-02-27:
Hello,

It's not going to blink, just show you the values below old_snr_alarm.

The default behavior was inspired by the TBS FPVision OSD fields. Remember that SNR is not a "level" or "power" like signal or battery levels. It is a ratio between noise level and received signal level. SNR is not "low" or "high". The purpose of SNR is to tell you how strong is the received signal compared to noise floor. Think of it as the "Check Engine" light in your car. Some "useful" values to remember are these:  

6dB = Received signal is 4 times stronger than noise. 4:1
3dB = Received signal is 2 times stronger than noise. 2:1
0dB = Noise and Received signals are the same. 1:1
-3dB = Noise is 2 times stronger than the received signal
-6dB = Noise is 4 times as strong than the received signal
-10dB = Noise is 10 times stronger than the received signal

You could also find out the RX's LoRa spreading factor your rfmod is using and look at the lowest SNR level the Rx could demodulate.

Without going into details, any value higher than 5dB, according to Semtech and my tests, is not useful or reliable and should not be considered. In fact, the value returned is for only one packet and currently the number returned by INAV to the OSD is filtered (planning to revert to the original code as soon as I get the configurator master to work!) so the only good number to look at is when snr is 0dB because below that your RX may be unable to demodulate and lose signal.

malinhiles, 2024-02-27:
Copy that, I understand the motivation there now. Thanks!

With this info as reference, should I also be weary of high RSSI values being inaccurate, since High SNR is unreliable? Or is it the reading of the noise floor that makes reported SNR unreliable at high values?

OptimusTi, 2024-02-27:
> Copy that, I understand the motivation there now. Thanks!
> 
> > SNR is a ratio between noise level and received signal level. Without going into details, any value higher than 5dB, according to Semtech and my tests, is not useful or reliable and should not be considered.
> 
> With this info as reference, should I also be weary of high RSSI values being inaccurate, since High SNR is unreliable? Or is it the reading of the noise floor that makes reported SNR unreliable at high values?

No, RSSI is good. Again, the radio chip in evaluating one packet of data. It's just chip limitations.

Don't lose sleep over it. Set it close to 0 and watch when it goes below that. That's the important info you get from it. How low can it go below 0dB before losing the ability to demodulate? You need to test. (again, don't rely on the osd. The value shown is filtered. use radio logs)

malinhiles, 2024-02-28:
@OptimusTi Gotcha, i see. Good to know. Thanks!