iNavFlight/inav #9725: When using self-compile firmware, unable to save mixer.

yxc407, 2024-02-15:
I flash the firmware I compiled into the flight controller, everything is fine, except that I cannot modify and save the mixer section. I tested three flight controllers, even if my firmware is not modified, simply compiled and flash, it is the same.

sensei-hacker, 2024-02-16:
Probably timer problems. Which target, and where are your changes?

yxc407, 2024-02-16:
OK,let me check,thank you

yxc407, 2024-02-16:
> Probably timer problems. Which target, and where are your changes?

I am attempting to make modifications from within the code.
[image]

This is the result I obtained after flashing.
[image]
[image]

My flight controller is JHEMCUF405, the target is JHEMCUF405, only modifying the code in target.c.
IMU and Baro are both displaying and receiving data normally, only the mixer cannot be modified and saved.

rizacelik, 2024-02-16:
Replace **TIM_USE_MOTOR** with **TIM_USE_OUTPUT_AUTO**

yxc407, 2024-02-17:
> Replace **TIM_USE_MOTOR** with **TIM_USE_OUTPUT_AUTO**

Sorry, I didn't explain clearly. I didn't originally modify this part of the code. This is the result after my modification, which is the same as the result before the modification.

sensei-hacker, 2024-02-17:
There is no command called "mixer", so that's why you're getting the error that's an unknown command.

There's mmix (motor mix) and smix (servo mix).

As Rizacelik said, that should generally be TIM_USE_OUTPUT_AUTO, whether you modified it or not.

closed 2024-02-18 by DzikuVx