iNavFlight/inav #9729: Standard Remote ID support for flying in the USA and EU

voxtelnismo, 2024-02-17:
## Current Behavior

Currently INAV does not support Standard Remote ID (per FAA definition).

## Desired Behavior

INAV supporting an external board-powered device, communicating with FC over I2C or UART, that broadcasts a programmable ID over WiFi or Bluetooth, compatible with FAA-expected protocol.

## Suggested Solution

Using an ESP32 or similar devboard, send GPS coordinates at the arming time, and transmit GPS coordinates IAW FAA MOC at the frequency of 1Hz. Upon loss of communications with the ESP32, immediately go into RTL mode.

## Who does this impact? Who is this for?

This is for all INAV users in the USA and EU who fly drones weighing over 250g.

## Additional context

There is a similar system already in use in Ardupilot repo.

MrD-RC, 2024-02-17:
IMHO, INAV should not do anything for remote ID. Aside from the fact that is it complete nonsense and does nothing for safety (in fact it makes it less safe for RC pilots). The last I heard a basic GPS coordinate does not meet the requirements for RID.

OptimusTi, 2024-02-18:
The recommended solution will never work. Remote ID, at least in the US, requires a device approved by the FFA.

sensei-hacker, 2024-02-18:
I don't personally care for the law, and wrote to the FAA about how it should be changed. I don't like the way the US Congress responded January 2015 event, but that can't be changed.

Making it more expensive for INAV pilots to follow the laws won't make the laws go away.

Similarly, encouraging users to switch to Ardupilot won't change the laws.

OpenDroneId is developed in accordance with the Means Of Compliance published by ASTM and accepted by the FAA. It is also designed to be compliant with similar regulations in Europe and Japan.

Again, I would like to change these regulations, but paying more for a proprietary module, and therefore carrying an extra GPS, doesn't make the regulations change.  Getting regulations changed requires other action.  FTCA continues to organize efforts to get the regulations changed. Making it more expensive by refusing to work with the open standard is not beneficial to those efforts.

Fracturing the open efforts only makes change *more* difficult. To make a credible case that the open community can operate safely through community -based efforts, we need to work together. The open standard is part of that.

sensei-hacker, 2024-02-18:
> The recommended solution will not work. Remote ID, at least in the US, requires a device approved by the FFA.

The FAA _has_ approved the MOC used by OpenDroneId.

DzikuVx, 2024-02-18:
This is not part of INAV project. INAV provides all required telemetry data over the MSP protocol and external board can get information like heading, coordinates, altitude, etc via MSP. 
However, INAV is not responsible for the additional hardware that can transmit this data.

closed 2024-02-18 by DzikuVx