iNavFlight/inav #9731: Unable to disable autoland if cruise is enabled on INAV7.1 RC1

korneliuszm, 2024-02-18:
Hi

During cruise flight I enabled RTH, as expected my wing came near home position and started autoland procedure. Few meters above ground I changed my mind and disabled RTH, but my autoland procedure was still enabled what caused further descent. I couldnt take control over the wing until I switched off cruise mode.

In my opinion when I switched off RTH, autoland also should be aborted.

## Steps to Reproduce
1. Enable RTH during cruise flight
2. Disable RTH during autoland

## Expected behavior
Autoland should be aborted

## Suggested solution(s)
Abort autoland when RTH is switched off

b14ckyy, 2024-02-19:
Hi @korneliuszm 
do you maybe have a DVR or log of the flight? 
To abort Autoland you normally just have to move the pitch/rill stick. The plane will then abort the landing and go back into RTH circle. 
How are your modes set up? are cruise and RTH controlled on separate channels so both can be on at the same time?

Scavanger, 2024-02-19:
Confirmed:
After cancelling, the old landing procedure was activated by mistake.

Choolet, 2024-02-21:
I have two problems.

The first is with my primary safehome. 
I always have a safehome set 300 meters from my launch area. This is because my club does not permit loitering over the pits.
If I place a safehome on the landing strip to only allow landing in a failsafe.  `nav_rth_allow_landing = failsafe`.
 It does not give me the option to use my primary safehome 300 meters away, when I select "safehome mode = RTH".

My landing strip is about 2 meters higher at one end than the other. This causes the aeroplane to crash into the ground when approaching from the high side of the strip. And if I set the approach and glide altitude higher to clear it. The aeroplane stalls above the runway.
Is there a way to set the inclination or declination angle of the landing zone on the final approach?

b14ckyy, 2024-02-21:
@Choolet this is a support question that either belongs to https://github.com/iNavFlight/inav/discussions or we can discuss in the FB groups or on Discord.

b14ckyy, 2024-03-20:
Fixed in https://github.com/iNavFlight/inav/pull/9794

closed 2024-03-20 by b14ckyy