iNavFlight/inav #9733: INAV 7 VTOL: MMIX cant saved

Zeppelin500, 2024-02-20:
## Current Behavior
I use a **Matek F405 VTOL** and a **FIMI Manta tilt rotor VTOL**
The "Motor 1" is connected to the motor at the rear.
In **tricopter** mode, everything work fine.
In **flying Wing** mode the Motor 1 should be off.
But if I try to save **mmix 0  0.000  0.000  0.000  0.000** at the GUI, the line is deleted and the settings of Motor number 2 becomes number 1 ( ...and Motor 3 becomes 2). 

## Steps to Reproduce

try to save:
mmix 0  0.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000
mmix 2  1.000  0.000  0.000  0.000

The result is in GUI:

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

In CLI all mmix are deleted.

## Expected behavior

mmix 0  0.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000
mmix 2  1.000  0.000  0.000  0.000

## Suggested solution(s)

Enable saving of "mmix 0  0.000  0.000  0.000  0.000"

## Additional context

# [https://pastebin.com/8cFZqW9y](https://pastebin.com/8cFZqW9y)

---
# version
# MATEK F405 VTOL aka -->
# INAV/MATEKF405TE_SD 7.0.0 Dec  5 2023 / 10:31:33 (895a4f31)
# GCC-10.3.1 20210824 (release)

DzikuVx, 2024-02-21:
```
mmix 0 0.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000
mmix 2 1.000 0.000 0.000 0.000
```
is not a valid mixer. Throtthe authority `0.00` are ignored

closed 2024-02-21 by DzikuVx

sensei-hacker, 2024-02-21:
Which means, make it 0.001 :)

Zeppelin500, 2024-02-21:
Thank you sensei. I already tried 0.005 (0.001 is also not storable) but the motor spins with 5%.
But I found a solution. -1.000 works!
My opinion was, motor1 at the rear is the tricopter default of INAV, not my choice. All tricopter VTOLs should have this Problem. If the Motor 1 in the front per default, there were no problem.