iNavFlight/inav #9737: The ground station cannot control the switch of the external buzzer

YangChen1234, 2024-02-21:
Current Behavior:
The alarm of the motor has been turned off, I use an external buzzer, normally the buzzer will be turned off when the BEEPER is turned off, and the buzzer will sound when the BEEPER is blue, but now the buzzer cannot be turned off after the power is powered on, I don't know what the reason is.

Below is a video of the phenomenon：

https://github.com/iNavFlight/inav/assets/89179705/52377b94-7928-44fa-9fdf-33b0bea5cd67

Jetrell, 2024-02-22:
The beeper mode slider is designed to turn ON or OFF the beeper in the case you are using it as a lost model alarm.. 
But that mode switch will not override warning beeps, caused by a hardware failure or status warning. Only the beeper mute can do that.
 This [doc](https://github.com/iNavFlight/inav/blob/master/docs/Buzzer.md) shows how to enable or disabled individual warning beep tones.

YI-BOYANG, 2024-02-25:
我猜现在这个声音是设备故障的警告声，而不是蜂鸣器控制的，你要检查一下传感器图标有没有标红

lida2003, 2024-02-25:
> 我猜现在这个声音是设备故障的警告声，而不是蜂鸣器控制的，你要检查一下传感器图标有没有标红

https://github.com/iNavFlight/inav/issues/9660

https://www.youtube.com/watch?v=A1eBqc499fE

YangChen1234, 2024-02-27:
It was an alarm .Thank you very much for your help. I have solved my problem with your professional guidance and advice.I change I was able to control the hardware buzzer by entering the relevant commands

closed 2024-02-27 by YangChen1234