iNavFlight/inav #9741: Accelerometer support on SpeedyBee F405 Mini target

Vins1417, 2024-02-21:
## Current Behavior
The accelerometer is well calibrated but in the setup tab, the values ​​move while the quad is stationary and no vibrations could disturb it. 

I noticed this because when I activate poshold mode, the drone shakes in all directions and I think that has a link with that.

## Steps to Reproduce
Flash Inav 7.0.0 on SPEEDYBEEF405MINI target ( https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/ )
Calibrate the accelerometer
Return to the setup page and look at the roll and pitch values ​​which are moving.

## Expected behavior
When the card is still, the pitch and roll values ​​of the accelerometer in the setup page should be there too, right ?

## Suggested solution(s)
I saw in the CLI (get acc) that there is a variable which defines the type of hardware (acc_hardware) and that's when I realized that the FC SpeedyBee F405 Mini is equipped with an ICM42688P accelerometer and that in the possible choices, it is not available. The base value for this target is ICM42605.. 

Wouldn't there simply be a compatibility problem?

## Additional context

I will attach a video of this problem

https://github.com/iNavFlight/inav/assets/148378237/fbd3ee50-4762-4614-afd1-090c187cfa5f

I tried with and without the magnetometer, and connected to the PC and via bluetooth on the living room table to be sure that there was no micro vibration from the computer for example.. I also tried with another FC Mamba in the same place under the same conditions and I don't have this bug with another FC Mamba

https://pastebin.com/k1kct21D

- FC Board name and vendor: SpeedyBee F405 Mini (https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/)
- INAV version string:
# version
# INAV/SPEEDYBEEF405MINI 7.0.0 Dec  5 2023 / 11:04:12 (895a4f31)
# GCC-10.3.1 20210824 (release)

Jetrell, 2024-02-22:
>When the FC is motionless, the pitch and roll values ​​of the accelerometer in the setup page should be motionless too, right ?

This is normal if your have a satellite fix... By the look of it, the green PPS light is flashing, saying its has a 3D fix ?
When this is the case. INAV's AHRS will actively be making heading and attitude adjustments to the gyro values, based on GNSS position data. Even when its sitting still.

Vins1417, 2024-02-22:
Yes you are right forgive me 🙏 I didn't know that.. I actually tried without having a GPS fix and the values ​​are immobile.. so my poshold problem comes from elsewhere 🤔

Vins1417, 2024-02-22:
Would it be wise to add the ICM42688P driver to the target or is the ICM42605 driver 100% compatible with the 42688?

Phil-MC, 2024-02-22:
I was helping with this on the FB Forum.  The gyro driver is missing from the TARGET files and needs to be 'included'. It is in the WING version but not in the MINI SPEEDYBEE F405.

mmosca, 2024-02-22:
> Would it be wise to add the ICM42688P driver to the target or is the ICM42605 driver 100% compatible with the 42688?

The driver in inav is the same. There are a few differences between the two but INAV handles it internally, even if it shows up as ICM42605.

You can also try 7.1 RC1 as there were changes to the driver that should improve performance of the ICM42688P imu.

Phil-MC, 2024-02-23:
Just for compatability with the WING version it could have my suggested entry. I don't think there's any issues with the WING.

Phil-MC, 2024-02-23:
There's quite a long discussion about this on FB FORUM  under a post from Vins Courtet.

The FC gyro was OK on Betaflight with the driver I mentioned, but on INAV it was not available and resulted in unstable readings.

Speedybee have obviously changed their component selection. I checked out the WING version of the FC and the TARGET files included the required driver. Hence my earlier comment suggesting that driver.

mmosca, 2024-02-23:
> I was helping with this on the FB Forum. The gyro driver is missing from the TARGET files and needs to be 'included'. It is in the WING version but not in the MINI SPEEDYBEE F405.

Are we talking about the wing mini or a different mini fc? The wing mini uses the F405WING target.

Vins1417, 2024-02-23:
> > I was helping with this on the FB Forum. The gyro driver is missing from the TARGET files and needs to be 'included'. It is in the WING version but not in the MINI SPEEDYBEE F405.
> 
> Are we talking about the wing mini or a different mini fc? The wing mini uses the F405WING target.

Not the wing mini.. the FC SpeedyBee F405 Mini 35A 20x20mm with the SPEEDYBEEF405MINI target.

https://www.speedybee.com/speedybee-f405-mini-bls-35a-20x20-stack/

mmosca, 2024-02-23:
I would recommend anyone using icm426xx gyros to try 7.1 RC1, as it is a lot closer to how BF and AP use this gyro.

Vins1417, 2024-02-27:
I just did a test, with version 7.1, basic settings for a 5" and it's day and night with version 7 🤩 there is indeed much better support for the ICM42688P gyro.
Thank you for your support 😍

closed 2024-02-27 by mmosca