iNavFlight/inav #9747: SPEEDYBEE F405V3 stopped working with inav

vasilisaaue, 2024-02-25:
I flashed the flight controller via inav and everything became normal. For several days the drone just stood on the shelf, I decided to connect it to the inav, but it did not connect. opens a connection and closes after a few seconds. Reflashing and completely erasing the chip did not help. I even flashed the drone on Betaflight and it worked. after which I flashed the drone again in INAV, but it no longer works. Help me please! There is no SD card inserted into my flight controller!

mmosca, 2024-02-27:
What version inav and configurator?

What operating system?

vasilisaaue, 2024-02-27:
> What version inav and configurator?
> 
> 
> 
> What operating system?

Inav 7.00 . Windows 11.

mmosca, 2024-02-28:
try disconnecting any devices on uart1 and 3 to see if it makes a difference

closed 2024-03-15 by mmosca