iNavFlight/inav #9752: CRITICAL SECURITY ISSUES: pre-arm ignored after quick disarm-arm sequence

sdellava, 2024-02-29:
## Current Behavior
Pre-Arm check is ignored if arm happen before 250ms after disarm.

**This is a very critical securiry issue because an involuntary re-arm can be fired while the pilot asked to disarm.**

## Steps to Reproduce

1. set the pre-arm check
2. pre-arm
3. arm
4. quick disarm and arm again. The FC arm even if the pre-arm switch is not in correct position.

## Expected behavior
if the pre-arm is set, after every disarm the pre-arm switch have to be verified before arm.

## Suggested solution(s)
I suppose the issue comes form the delay introduced to correcly save data in the bb. This has been obtained adding a 250 ms of delay from the disar request to the disarm execution.

I suggest two logically split the disarm process in two step. In the first step, motors are immediatelly stopped and the arm status is set to off, then after 250ms the rest of the disarm process is completed. 

If the arm status is off, then the pre-arm is required to re-arm again (if pre-arm is set).

breadoven, 2024-02-29:
I assume you're talking about `switch_disarm_delay` which as I understand it is related to failsafe issues. It's intended to prevent a false disarm when the RC signal is unreliable which could lead to the Arming channel doing odd things (switching on/off erratically). If it is this setting then it can be reduced to 0 although this would increase the chances of an unwanted disarm I guess.

b14ckyy, 2024-03-01:
@sdellava just for my understanding: In what situation could someone Disarm and rearm within 250ms by accident, so it could pose a thread? Can you give a practical example? 

the disarm delay was introduced mostly for FrSky ACCST V1 and Pre V2.1 firmware versions that tend to glitch on the first of 2 received channel data packages and moved half the channels to 1000us briefly. Possibly also other receivers that use a cloned ACCST V1 protocol like Radiomaster or Jumper clone receivers.

sdellava, 2024-03-01:
> @sdellava just for my understanding: In what situation could someone Disarm and rearm within 250ms by accident, so it could pose a thread? Can you give a practical example?

Certainly. This was discovered by accident by testing with a drone. The pilot wanted to disarm but accidentally immediately rearmed. He didn't realize he had rearmed, although he realized it shortly after, but he was surprised anyway because he assumed he couldn't accidentally rearm since he had configured pre-arm.

Therefore, this is not a use case, but a potential risk that may lead to not disarming not disarming promptly  because of an error by the pilot accidentally rearming.

Regarding the @breadoven observation about the risk of failsafe, on second thought I think that yes, the delay makes sense, but only if a failsafe aprocedure is not set up. In that case anyway there is no risk of losing control in the absence of a signal because it still does not disarm but performs the failsafe procedure with its eventual delay.

sdellava, 2024-03-01:
I think a solution could be found by distinguishing the absence of a signal from the receipt of a disarm signal.

The disarm signal is necessarily explicit and being an emergency command should never be delayed much less ignored.

Different is the case where, in the absence of radio link and in the absence of a defined failsafe procedure, then the FC autonomously activates the disarm. In this case it makes sense to wait because obviously the signal can come back.

harryjph, 2024-03-02:
I personally set `switch_disarm_delay` to 0 on modern radio equipment.

However, the default is currently 250ms and this should probably be closed as intended behaviour.

MrD-RC, 2024-03-02:
I agree. This is not a bug as it is expected behaviour. The fact that only 1 person has reported it as an issue in  3 years shows that it is very much an edge case. Though it may be worth documenting somewhere. 

It is also be mentioning that if they disarm in flight. They can also re-arm, just with the arm switch, in 5 seconds. Perhaps the quad thought it was still flying at this point. So there is a possibility that this is not related to #6737.

sdellava, 2024-03-02:
I agree with all you guys. Setting the delay to zero seems the best logic. 

So I'm closing this issue. But I guess a better documentation is needed because the final behaviour of the drone is the result of many "delays" and the failsafe procedure, not only the switch_disarm_delay.

closed 2024-03-02 by sdellava