iNavFlight/inav #9755: [7.1 RC1] Sensor funsion: Altitude going nuts in Realflight/SITL when Baro is used

Scavanger, 2024-03-01:
Sensor fusion does not work properly in SITlL, at least in RealFlight, XPlane not yet tested.

If the barometer is used, the altitude calculation starts to go crazy and displays seemingly random altitude values, alternately too high or too low, which causes extreme dolphining in the NAV modes.
If the barometer is switched off, everything is normal, which of course has no practical effect in SITL, as the GPS data is very accurate and does not fluctuate. 

At first I thought it was the magnetometer, as that fixed the error for me recently, but it seems to be the barometer.

b14ckyy, 2024-03-20:
Fixed I guess? https://github.com/iNavFlight/inav/pull/9724

Scavanger, 2024-03-20:
Nope, unfortunately not.
I thought that was the error, but randomly the altitude is sometimes correct, but in most cases it jumps up and down randomly.
The error must be somewhere in the new sensor fusion code.