iNavFlight/inav #9759: Failsafe mode control

uranus36, 2024-03-02:
## Current Behavior

If I choose RTH for failsafe, it would execute failsafe command aways.
## Desired Behavior

When my FW is still on the ground , I would prefer the failsafe command been changed to `drop` at the meantime. Or my plane could take off immediately. What if somebody or something is just right in front of my plane, it could be dangerous.
## Suggested Solution

Add a failsafe choose mode in the Mode tab.

Add a failsafe choose mode in the Mode tab.
## Who does this impact? Who is this for?

It's for all airplane users. Just in case that the plane could take off unexpectedly.
## Additional context

MrD-RC, 2024-03-02:
It should not just randomly execute failsafe. If you are on the ground, the aircraft should be disarmed. You should also never power off the transmitter before powering off the aircraft. 

There are edge case instances where an aircraft has landed below a crest or other blocking mass. Which has caused a failsafe. But this is extremely rare.

uranus36, 2024-03-02:
Thanks for your reply. But I just encounter a very extreme case . 3 days ago, when I landed my FPV T28 to the ground and drove it to the taxi way. Suddenly, FC triggered FS and the plane stared to t take off with full throttle. Fortunately, there's no one stand on the taxi way. I checked my TX16S and found everything looks normal including my TBS transmitter is still running with 500mW. But I can't get any feedback from stick movement. Thus , I can only stand there and watch my plane to fly circle for couple turns and dropped to the ground while it's out of battery. I'm still trying to figure out what's wrong with it. But , if there's a failsafe switch  , I would like to switch it  to 'drop' mode. Then, it should be able to reduce the danger in taxi way . Please consider to add the function in the future. 
Besides, while the failsafe RTH is executing ， can I set the gear to be up ?

Jetrell, 2024-03-03:
@uranus36 Have you tried these settings ? 

[image]

uranus36, 2024-03-04:
> @uranus36 Have you tried these settings ?
> 
> ![FS](https://private-user-images.githubusercontent.com/47995726/309521389-fa65ff88-4e32-45c8-85e2-6ade516dd687.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1MTkzNzksIm5iZiI6MTcwOTUxOTA3OSwicGF0aCI6Ii80Nzk5NTcyNi8zMDk1MjEzODktZmE2NWZmODgtNGUzMi00NWM4LTg1ZTItNmFkZTUxNmRkNjg3LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA0VDAyMjQzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5NjVmNWIyYWVjM2YxOGUyYWI2ZDEwZjY2ZDExZGJkODI0OGFhYWVkNTViNmZjYmVmZTczMWFlNzNhYzk0ZTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.V3VhgkNiWNv-sb2uluCD51kxVj4RtkODurEyIlVdVWI)

Hi I did setup the minimum distance for 5 meters. And some of the setting is still not available in my INAV 5.1 . Besides , I setup a safehome position which is about 200meters in front of me , the lost communication position was just about 30 meters in front of me . Once the FS is triggerd , it should still take off to the sky per RTH request.
Thus , I would like to have some suggestion as below:
1. Add a failsafe action tab for perhaps I need to put the gear up when it's FS
2. Add cruteria to check current height to confirm RTH or drop need to be executed
3. Add a switch to control RTH or drop need to be executed now

sensei-hacker, 2024-03-04:
If you want it to land 200 meters in from not of you, you may want to increase the alternate distance from 5 meters to maybe 100 or more.

Unfortunately, height relative to home isn't all that useful unless the aircraft is at the point, because people regularly fly from hilltops, building roofs, etc where they get a good view of the surrounding area. 5 meters higher than the home point may be 200 meters above the ground.

MrD-RC, 2024-03-04:
1. Setting up landing gear up is something you can already do using the programming tab in configurator. Undercarriage. This doesn’t need a specific failsafe setting.
2. Altitude is relative and can drift. So this is not a good thing to base any critical features on. 
3. A switch for RTH/Drop would be pretty useless if this is happening in failsafe. You’d have no RC link, so would be unable to activate the switch. 

When you had this issue:
- Could you see your aircraft?
- Was it obstructed?
- Was it a hard landing or crash?
- When tested back at the pits, did the RC link work?

If the above answers are yes, no, no, yes. Then you really need to look at your antenna installation. If you have landed normally, and don’t lose sight of the aircraft. It should have never entered failsafe in the first place.

b14ckyy, 2024-03-22:
After reading here, and I agree the main culprit is probably the RC link itself and a config issue (drop distance), I wonder if it makes sense to add an extra safety layer. 
Maybe not so useful for quads as you usually just disarm them anyway when landed and you land near you, but for fixed wings that might land a bit more off site. 

What if we implement a "Landed"-Flag? 
when taking off, over a certain speed threshold and/or a certain altitude threshold (lets say 5m) the landed flag is removed.
For landing, we have the landing bump detection that is capable of disarming the craft but what about using this to set the LANDED-Flag back on if auto disarm is not used? Then the landed flag stays on until the speed threshold for takeoff is once passed below and then it has to accelerate again before it is cleared again. 

as long as the LANDED flag is active, default failsafe action should always be dropped. This would prevent runaways if the RC link fails after landing.

b14ckyy, 2026-09-22:
Closing as fixed in #10135

closed 2026-09-22 by b14ckyy