iNavFlight/inav #9761: VSpd Telemetry value not refreshing in 7.1 Rc1

jureopacak, 2024-03-02:
## Current Behavior
VSpd telemetry value (which I use for Variometer) is not working in 7.1 Rc1.

Under INAV sensors I see Baro data and that seems to work perfectly fine but looks like it is not pushed as telemetry data. See attached Image.

[image]

## Steps to Reproduce
1. Have Inav 7.1 RC1
2. Have ELRS (I use Radiomaster RP1 with Ranger Micro), EdgeTX "Providence" v2.9.4 
3. Check Telemetry values. In my case VSpd is not refreshed.

## Expected behavior
VSpd telemetry value should always be there if Baro sensor is working.

## Additional context
- I noticed that value in the last flight was not 0.0/NA but stuck to some value like -0.4 if I remember correctly. If needed I can check telemetry logs from my remote.
- **After this behavior is noticed I reverted to INAV 7.0 and VSpd was working as expected. I used same Diff file as for 7.1 Rc1 (without auto landing settings).**

---

- FC Board name and vendor: SpeedyBee F405 Wing
- INAV version string: INAV/SPEEDYBEEF405WING 7.1.0 Feb 14 2024 / 15:58:19 (49325426)

b14ckyy, 2024-03-20:
can you validate that this is not an ELRS issue? 
ELRS on latest version? Tried higher telemetry rate? Anyone else you know with a different RC system can check if they see the same issue?

stronnag, 2024-03-20:
Maybe you should take into account that CRSF does not provide Vspd ....

jureopacak, 2024-03-20:
Downgrading to 7.0 resolved the issue. So both, 7.1 when it did not work and 7.0 when it worked are tested on exactly same setup.
To verify this I flashed again 7.1 and telemetry value didn't refresh.

ELRS Version I used is 3.1.1 CE_LBT. With 500Hz 1:32 Telemetry ratio. Same for 7.0 and 7.1.

Once time allows I will try to do the same test with the recording.

b14ckyy, 2024-03-20:
OK can you please try with this build: [https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl](https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl&fbclid=IwAR0IlLXyWr2ukDHmgJ6uxVPwZ9ys8VTPHdSy6ijLCc0ST_gsbhiuSVEtcrE)

its also 7.1 but with latest patches and fixes implemented. If it happens there as well, someone needs to look at it.

jureopacak, 2024-03-20:
Ok, just tested again. Same hardware just different Inav version same diff. 

Basically Inav 7.0 has 24 Telemetry items but Inav 7.1 has 23 Telemetry items (just Vspd is missing)

## It works with 7.0

[image]

## It doesn't work with 7.1 RC1

[image]

## It doesn't work with 7.1 RC2

[image]

Please tell me if there is anything else I can test.

b14ckyy, 2024-03-20:
I informed the devs.

b14ckyy, 2024-03-20:
@r1000ru can you please have a look if this has to do with https://github.com/iNavFlight/inav/pull/9438 ? 
did you test this with EdgeTX?

r1000ru, 2024-03-21:
@b14ckyy, I'll see what the problem is. I assume that this is due to the fact that GNSS has an advantage over baro, but indoors it does not. I’m not at all happy about the forced assignment of such an advantage. This can generally be used on gliders without a FPV, but with a FC, when you actively use the variometer? Although maybe I'm wrong.

sensei-hacker, 2024-03-21:
@r1000ru  It seems to me, if you were receiving vario data that was more weighted to GPS, you would be receiving vario data. :)   Since you aren't receiving the value, I don't think it's caused by receiving a value that weights GPS more.

I think the fact that the value was moved in commit 5d77d07ce is more likely the reason you are having trouble with it.  It is now combined with the baro message.  EdgeTX supports this since 2.8.

Please double check with "discover new" to see if your radio discovers the new baro/vario message. If not, there is an incompatibility or issue between the way INAV is sending that and the way EdgeTX is receiving it.

r1000ru, 2024-03-21:
@sensei-hacker, I did the support for the baro height and the variometer in one packet 0x09. The baro height is taken directly from the sensor. The value of the variometer is taken after many calculations. For the barovariometer packet, I will get the variometer value from the data from the barometer. For a separate variometer package, this is how INav calculated this value taking into account GNSS data. According to the screenshots of the creator issue, we can see that at the time of the test, he had no GNSS fix.

jureopacak, 2024-03-21:
@r1000ru fyi maybe helps. Yesterday when I tested it (screenshots) I was indoor and had no GNSS fix, but when I noticed Vspd is not working I was on the field flying, GNNS fix was there. If needed I can upload EdgeTx logs from that day when I was flying.

MrD-RC, 2024-03-21:
> @sensei-hacker, I did the support for the baro height and the variometer in one packet 0x09. The baro height is taken directly from the sensor. The value of the variometer is taken after many [...]

If there’s no GNSS fix. Surely pure barometer should be used? It’s more accurate than GNSS altitude after all.

b14ckyy, 2024-03-21:
I am not sure if its the GNSS/Baro priority. 
That would not explain why Vspd is missing completely in the latest 7.1 build from 2 days ago. there must be another commit between RC1 and my inofficial RC2 that interferes with the sensor telemetry in addition.

r1000ru, 2024-03-21:
@jureopacak , Are you using EdgeTX 2.10.0RC1? I made changes to the INAV code but didn't get the result. I have now looked at the EdgeTX code, they have stopped processing the value of the variometer in the barovario packet. Before that, there could have been two variometers (package 0x07 and package 0x09)

jureopacak, 2024-03-21:
I am using EdgeTX "Providence" v2.9.4

b14ckyy, 2024-03-21:
Interestingly in antother telemetry related issue, I asked the poster, to test it on his setup. Also ELRS and EdgeTX and he has no issues with 7.1. His problem was the outdated Module Firmware as it seems: 

https://github.com/iNavFlight/inav/issues/9685#issuecomment-2012043881

r1000ru, 2024-03-21:
@jureopacak, @b14ckyy ,I made a mistake with the data about edgetx. Previously, GPS altitude sensors and altitude sensors were named the same. As for the transmission of variometer data in the barometer packet 0x09, this data overwrites the variometer data from the individual packet 0x07. Without EdgeTX refinement, we cannot output two Vspds. 
Well, the main question is whether it is necessary. Since it was decided that the GNSS data has priority over the barometer data, I see no point in going back.

b14ckyy, 2024-03-21:
I just wonder, isn't Vertical Speed a value that gets spit out by the Sensor fusion anyway, no matter what is used, GPS or Baro or both? The Sensor fusion should do all that merging and then spit out a final vertical speed value, right? 

I am a programming noob, I might be wrong about that. But then why even bother reading raw data?

b14ckyy, 2024-03-22:
If we cannot fix and test this in time, please let's roll back the change that broke this functionality so we can go forward with the 7.1 release. Let's move that change to 8.0 then. 
Pawel wants to build 7.1 next week so we can release next weekend and before eastern. 
This is the only current showstopper bug as it will break the vario for glider pilots.

sensei-hacker, 2024-03-22:
At minimum, let's put back the old variometer packet, in addition to the new value.

r1000ru, 2024-03-22:
The sensor has nothing to do with it and works fine. To return to his previous job, it is necessary to lower the priority of GPS over baro in console.

b14ckyy, 2024-03-22:
The problem is, that in RC1 he does not see the sensor at all. It does not even show up in the sensors list when discovering sensors. So something is missing there.

Jetrell, 2024-03-23:
If it helps to know. I pulled out my old OpenTX radio. With the last release of OTX. And tested it with TBS CRSF and INAV 7.1.. The VSpd doesn't show with it either.. But it does with 7.0 and before.

mmosca, 2024-03-26:
Does it behave correctly in edgetx?

b14ckyy, 2024-03-26:
@mmosca this is EdgeTX https://github.com/iNavFlight/inav/issues/9761#issuecomment-2010532488

b14ckyy, 2024-03-26:
@jureopacak confirmed, all back to normal after rollback of https://github.com/iNavFlight/inav/pull/9438 
So we will roll back for the 7.1 release to keep Vspd working and then it can be figured out for 8.0 if that change makes sense and is properly tested with different setups.