iNavFlight/inav #9763: FW Auto Land inadvertent disarm during FS RTH

breadoven, 2024-03-02:
FW Auto landing disarms inadvertently before landing when FS RTH is active because the following drops out to `RTH_idle `causing an instant FAILSAFE landing and disarm to occur in the Failsafe module:

https://github.com/iNavFlight/inav/blob/622b6010f79a2136a55f391aff4216fb9978d833/src/main/navigation/navigation.c#L4893
https://github.com/iNavFlight/inav/blob/622b6010f79a2136a55f391aff4216fb9978d833/src/main/flight/failsafe.c#L537

Caused by the flight mode changing from RTH to WP mode when switching to `NAV_STATE_FW_LANDING_APPROACH`:
https://github.com/iNavFlight/inav/blob/622b6010f79a2136a55f391aff4216fb9978d833/src/main/navigation/navigation.c#L1096

In fact it's not good the way the FW Auto Landing phases change flight modes (mapToFlightModes) during the different Auto landing phases without actually switching to that flight mode. The flight mode flags affect various things in unpredictable ways if you're not actually in that mode.

- INAV version string: 7.1 RC1

Scavanger, 2024-03-05:
Thanks for reporting.

I have already fixed it and am currently testing it. 

The idea with the extra flight mode is good, I had it in at the beginning and then forgot about it in the end, but in many hours in the simulator and in the field tests I didn't notice any negative effects but better safe than sorry.

b14ckyy, 2024-03-20:
The fix should be in here? Merged now https://github.com/iNavFlight/inav/pull/9794