iNavFlight/inav #9776: Inav 7.0.1 doesn't work for Skystars F722HD Pro4 FC.

PredySoft, 2024-03-07:
## Current Behavior

 Recently I flashed INAV 7 Firmware to my Skystars F722 HD Pro4 FC. In INAV the target is SKYSTARSF722HD. I selected lates stable version and flashed firmware. But INAV doesn't detect anything except barometer so at the end it is not working at all. 

## Steps to Reproduce

1. Install INAV 7.0.1 on Win11
2. Plug USB cable to Skystars F722 Pro 4 FC.
3. Select target SKYSTARSF722HD
4. Select  Related 7.0.0 SKYSTARSF722HD version
5.Load Online
6. Flash the firmware.
7. After connect FC to INAV no sensors will be detected except Barometer.

## Expected behavior

All present sensors will be detected and cofiguration possible.

## Suggested solution(s)

No clue

## Additional context

---

- FC Board name and vendor: Skystars F722 HD Pro4
- INAV version string: 7.0.1

PredySoft, 2024-03-13:
Let me add one observation: When I use target SKYSTARSF722MINIHD instead of right one (SKYSTARSF722HD). Accelerometer, Gyro and other sensors are detected. FC is just detected flipped with head down so after alignment it seems to be working . That I would say indicates this is not a HW issue, but the bug in SKYSTARSF722HD target code or different sensors used on FC than declared by documentation, but I think MINI version is using the same sensors. Thanks.

b14ckyy, 2024-03-14:
Probably not supported target.
The BF target would be SKYSTARSF722HDPRO

If that's not in inav you would need to make your own.

PredySoft, 2024-03-14:
[image]
FC documentation says it is supported and SKYSTARSF722HD is correct one.

b14ckyy, 2024-03-14:
I see, my bad, I have not seen this image on my phone. 
Can you please try INAV 6.1.1 if that works? The target had some compatibility changes for 7.0 but no change prior to that. So if 6.1.1 works, then these changes might have broken something.

DzikuVx, 2024-03-14:
The `SKYSTARSF722HDPRO` will be release with INAV 7.1.
Manufacturer changed used baro hardware after INAV 7.0.1 was released.
My suggestio is to use unstable 7.1 INAV RC1

PredySoft, 2024-03-14:
Ok Thanks, let me try.

kondrat44, 2025-05-22:
Hello,

I am experiencing an issue with my Skystars F7 HD Pro 4 (30x30 mm) flight controller. After attempting a firmware update, the board stopped being recognized by the computer — it now shows up as an “Unknown device” when connected via USB, and only the red LED is lit on the board. I suspect the bootloader may be corrupted, and I need to restore the board using a programmer.

I have an ST-Link V2 (STM8/STM32) programmer. I would like to connect to the flight controller using STM32CubeProgrammer, erase the current firmware, and flash a new one to bring the board back to working condition.

I kindly ask you to explain in detail or, if possible, provide a connection diagram for how to connect the ST-Link V2 to the Skystars F7 HD Pro 4 flight controller. Specifically, I need to know:
	1.	Which exact pins on the ST-Link V2 should be connected to which pins on the flight controller?
	2.	Should external power be supplied to the board during flashing?
	3.	How to correctly configure STM32CubeProgrammer for connection and recovery?
	4.	Are there any special considerations when recovering this particular board?

Thank you in advance for your help and detailed instructions!

sensei-hacker, 2025-05-22:
> Hello,
>
> I am experiencing an issue with my Skystars F7 HD Pro 4 (30x30 mm) flight controller. After attempting a firmware update, the board stopped being recognized by the computer — it now shows up [...]

You can just use the DFU button on the board.
You can't accidentally overwrite DFU - it's still there.
Hold the DFU button while plugging in the USB cable.  Then flash the correct target with erase all checked, and "no reboot", since it will already be in DFU mode.

What may have happened is if the wrong INAV target is flashed, it won't be able to connect to *INAV*. But that doesn't affect DFU.

Theoretically it's *possible* to mess up the connection to INAV by turning off erase all while upgrading from a 5.x, 6.x, or 7.x to 8.x.  But still, that can't break DFU.

kondrat44, 2025-05-22:
My board doesn’t respond to the button and doesn’t enter DFU mode.

kondrat44, 2025-05-22:
Could you please explain to me where exactly I should solder the programmer to the board? I just can’t figure it out — DFU mode doesn’t work on any of our boards. I have six of them, and DFU mode doesn’t work on any of them. I don’t know why.

sensei-hacker, 2025-05-22:
At a glance, I don't see any programming pads marked on the diagram. Since you actually have the board in your hands, you can look closely, maybe with a magnifier app on your phone, for small pads that may be for that.

What may be more likely to help is to run the driver fixer so you have the correct DFU driver. That's very often a problem for Windows users.

kondrat44, 2025-05-22:
Вот улучшенная версия вашего текста на русском и его перевод на английский:

Улучшенный русский вариант:

С Windows точно нет проблем — она работает корректно. Однако ChatGPT упомянул, что можно восстановить плату напрямую с помощью программатора. После перепрошивки платы Windows перестала её распознавать и теперь при подключении через USB-C отображается как «неизвестное устройство». Ранее такой проблемы не было. Сейчас моя основная задача — определить, куда именно припаять контакты от программатора. Это единственное, что мешает мне восстановить плату. Пожалуйста, помогите с этим.

Перевод на английский:

There are definitely no issues with Windows — it works fine. However, ChatGPT mentioned that it’s possible to recover the board directly using a programmer. After I reflashed the board, Windows stopped recognizing it and now shows it as an “unknown device” when connected via USB-C. This issue didn’t exist before the reflash. My main problem now is figuring out where exactly to solder the programmer’s wires onto the board. That’s the only thing preventing me from restoring it. Please help with that.

https://fixfly.ru/files/f7hd-pro4-km65-spec.jpg

kondrat44, 2025-05-22:
There are definitely no issues with Windows — it works fine. However, ChatGPT mentioned that it’s possible to recover the board directly using a programmer. After I reflashed the board, Windows stopped recognizing it and now shows it as an “unknown device” when connected via USB-C. This issue didn’t exist before the reflash. My main problem now is figuring out where exactly to solder the programmer’s wires onto the board. That’s the only thing preventing me from restoring it. Please help with that.

https://fixfly.ru/files/f7hd-pro4-km65-spec.jpg