iNavFlight/inav #9782: iflight blitz atf435 flight controller

bwilliams1132, 2024-03-09:
inav wont read my accelerometer or gyro only barometer

KznProkuror, 2024-03-12:
Same problem, I went through everything in the sensors, it doesn’t want to find it(

KznProkuror, 2024-03-12:
> inav wont read my accelerometer or gyro only barometer

Did you manage to solve the problem?

bwilliams1132, 2024-03-12:
U got a way to fix it

bwilliams1132, 2024-03-12:
No I was hoping you did

Phil-MC, 2024-03-14:
The manufacturer might well have updated the chips on the board. It is only advertised for Betaflight so the INAV TARGET may well be out of date.

Phil-MC, 2024-03-14:
The advertised components are
Gyro: BMI270
Barometer: DPS310
OSD Chip: AT7456E
Flash: 32MB (Blackbox flash)
UARTS: 6

bwilliams1132, 2024-03-14:
Ok thanks for your help phil

Phil-MC, 2024-03-14:
As far as I can see from the TARGET . h the basic drivers are all covered. 
With a number of users reporting the same problem, it is unlikely to be a faulty board.

It needs someone with Betaflight on the board to check their devices (resources?) and see if they have been altered.

bwilliams1132, 2024-03-14:
I just got my stock betaflight cli dump....

# version
# Betaflight / AT32F435G (A435) 4.5.0 Aug 10 2023 / 07:39:06 (50da879b1c) MSP API: 1.46

# start the command batch
batch start

board_name IFLIGHT_BLITZ_F435
manufacturer_id IFRC

# name: Chimera7 Pro V2

# resources
resource BEEPER 1 B02
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B09
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C15
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
[... 999 lines cut ...]

# rateprofile 0
set rateprofile_name = DEFAULT
set thr_mid = 50
set thr_expo = 0
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 18
set pitch_rc_rate = 18
set yaw_rc_rate = 18
set roll_expo = 70
set pitch_expo = 70
set yaw_expo = 70
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
save

Phil-MC, 2024-03-14:
I looked but couldn't find any reference to the IMU device in your dump.

Phil-MC, 2024-03-14:
@mmosca Any ideas?

KznProkuror, 2024-03-14:
> @mmoscaЕсть идеи?

My sensors are also not detected, how can I help? I'll give you all the information

closed 2024-03-15 by mmosca

Phil-MC, 2024-03-15:
Thanks @mmosca for sorting it out.