iNavFlight/inav #9783: Internal baro and magnetometer does not work together

butcher51111, 2024-03-09:
If GPS sca scl pad is connected to fc the baro stops work, if sca scl pad is free, baro working again. 

I tested it on two different FC: Kakuta H7 , Foxeer f722 V4, both setup working well with betaflight.

MrD-RC, 2024-03-09:
There is something on the GPS (most likely the magnetometer) with the same address as the baro. You will either need to use a different I2C bus if available. Ot try a different GPS module.

butcher51111, 2024-03-09:
> There is something on the GPS (most likely the magnetometer) with the same address as the baro. You will either need to use a different I2C bus if available. Ot try a different GPS module.

Then how is it working flawlessly on BF?

sensei-hacker, 2024-03-09:
Which GPS unit are you using?

butcher51111, 2024-03-09:
> Which magnetometer (compass) / GPS unit are you using?

Ahh sorry, I forgot that: GEP-M10-DQ

sensei-hacker, 2024-03-09:
That's what I figured. That's the one that has it's own extra barometer. I suppose for use with the JB FC, which is missing a baro.

So you have two baros hooked up. INAV asks "hey barometer, what's the pressure?" And the baro answers "oneonehundredtthounhundredandonethouonethoudsand ..." because there are two baros talking at once.

butcher51111, 2024-03-10:
> That's what I figured. That's the one that has it's own extra barometer. I suppose for use with the JB FC, which is missing a baro.
>
> So you have two baros hooked up. INAV asks "hey barometer, what's [...]

Ohh shoot, I didn't notice that 🤦‍♂️. Nevertheless, for noobs like me it's really hard to figure out, somehow it would be nice to disable one or other since it's working with BF somehow. Anyway, thx for the help!

YI-BOYANG, 2024-03-11:
So which brao model is integrated on your flight control? Is it also a DPS310?

butcher51, 2024-03-11:
> So which brao model is integrated on your flight control? Is it also a DPS310?

Based on the description of kakute h7 has the BMP280, I couldn't find any info about the foxeer fc, but both behave the same way.

sensei-hacker, 2024-03-11:
> it would be nice to disable one 

Got a tiny soldering tip or hot air? And some tweezers? :). You could take the baro off that GPS board pretty easily.

INAV can't disable BMP280, because that's all it could do - ignore the BMP280. Two of the same sensor attached to the same bus are indistinguishable. Betaflight may have just ignored the baro, or gotten lucky, or whatever. BF doesn't care as much about getting an accurate baro reading and may be fine with a jumbled reading.

YI-BOYANG, 2024-03-11:
[image]
If you really need to use the GEP-M10-DQ, but want the barometer to work properly, then you can remove the red circle of this resistor, when removed it will turn off the DPS310 on the GPS, so that it will not affect the BMP280 on the flight control.
When you need to use the DPS310 again, you only need to short-wire this position with solder.

closed 2024-03-15 by mmosca

CatalinOAlexandru, 2024-10-14:
For anyone in the future finding this, I contacted geprc, and they gave me this information about removing a different transistor. I can confirm that for GEP M10 DQ, removing the transistor in my images worked.

[image]
[image]
[image]
[image]

The actual docs they shared with me.
[GPS集成的磁力计和气压计问题解决方案.docx](https://github.com/user-attachments/files/17369851/GPS.docx)

 Happy Flying!

nervetnova, 2024-10-30:
I'm experiencing an issue with INAV 7.1.2. The Mamba F722 MK1 flight controller does not have a barometer on the board, and this should not conflict with the barometer on the Geprc M10DQ, but the barometer icon is still showing red. The MAG and GPS are working well. I'm looking for a way to recompile INAV; can you help me. Thanks

YI-BOYANG, 2024-10-30:
> I'm experiencing an issue with INAV 7.1.2. The Mamba F722 MK1 flight controller does not have a barometer on the board, and this should not conflict with the barometer on the Geprc M10DQ, but the [...]

What is The target name of The Mamba F722 MK1 flight controller? I might be able to check.

YI-BOYANG, 2024-10-30:
I found that the target name of Mamba F722 MK1 is MAMBAF722_I2C. I checked its definition and found that DPS310 is missing. Now I have added DPS310 and recompiled the firmware, please burn this firmware and then check whether DPS310 works properly.
[inav_7.1.2_MAMBAF722_I2C(DPS310).zip](https://github.com/user-attachments/files/17570159/inav_7.1.2_MAMBAF722_I2C.DPS310.zip)

YI-BOYANG, 2024-10-30:
> I'm experiencing an issue with INAV 7.1.2. The Mamba F722 MK1 flight controller does not have a barometer on the board, and this should not conflict with the barometer on the Geprc M10DQ, but the [...]

I found that the target name of Mamba F722 MK1 is MAMBAF722_I2C. I checked its definition and found that DPS310 is missing. Now I have added DPS310 and recompiled the firmware, please burn this firmware and then check whether DPS310 works properly.
[inav_7.1.2_MAMBAF722_I2C(DPS310).zip](https://github.com/user-attachments/files/17570202/inav_7.1.2_MAMBAF722_I2C.DPS310.zip)

YI-BOYANG, 2024-10-31:
#10442

nervetnova, 2024-12-03:
> I found that the target name of Mamba F722 MK1 is MAMBAF722_I2C. I checked its definition and found that DPS310 is missing. Now I have added DPS310 and recompiled the firmware, please burn this [...]

Can you support it on Inav8?. thanks again.