iNavFlight/inav #9785: Matek F722 Wing: Barometer plus Magnetometer does not work

Slash7FPV, 2024-03-10:
Hi all,
if i connect the sda and scl of the magnetometer in Inav 7 (Walksnail M10 gps) the Barometer of the Matek F722 Wing FC is not recognized anymore. It does not matter if the Magnetometer is software side activated or not, it is enough to be hardwired to the FC to cause the baro to not working anymore.
Regards
Jan aka Slash7 FPV

**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Jetrell, 2024-03-10:
>It does not matter if the Magnetometer is software side activated or not, it is enough to be hardwired to the FC to cause the baro to not working anymore. 

What voltage source is driving your GNSS unit.. 5v or 3.3v.. Does the barometer continue to work if you disconnect the GNSS /magnetometer.. If so. It maybe be overloading the 3,3v LDOR. Or there could be feedback occurring between the two regulation systems. Shutting down the 3.3v supply..
I vaguely remember this issue with the Matek F722 wing. When I was using the i2C temp sensors. It would also cause the baro to shut down.

Slash7FPV, 2024-03-11:
Hi, it´s connectetd to the Uart 3 line on the Matek F722 Wing, so TX3-RX3-4v5-Gnd-SDA-SCL goes to the Walksnail M10 GPS.
The GPS module sits outside on the Wing because of HDzero interferences which gives cable lenghts of around 45-50cm. 
The Baro works only if i disconnect SDA SCL. Just disabling the Magnetometer in Inav does not help. 
GPS only works fine together with the Baro.
Voltage seems just fine and stable measured on all outputs and connected devices.

b14ckyy, 2024-03-22:
Turned out to be a Clash between the Walknsail used QMC Compass and the BMP280. Different GPS with IST8310 Compass works. Closing.

closed 2024-03-22 by b14ckyy