iNavFlight/inav #9790: arming with "yaw-right" gets the motors to spin instantly at full throttle

sgrizzi, 2024-03-11:
****

## Current Behavior
With reference to issue #9777 I did run some additional tests and used blackbox logging.
My quadcopter is using INAV 6.1.1 on a SpeedybeeF405V3 controller, running 4x 2207 motors with 5" propellers. The esc's are running BLHeli_S 16.7, as programmed by default on the F405V3. BE880 gps + magnetometer.

## Steps to Reproduce
throttle-idle = 18%
when I arm the quad using “yaw-right” two motors (motor 2 and 3) start instantly spinning at 83%.
I need to disarm to stop the motors.

throttle-idle = 17%
arming with “yaw-right” will arm the quad but the motors run on/off unstable - yet, when they run, the speed is normal.
Raising the throttle will raise the motor speed normally as expected.

throttle-idle < 17%
arming with “yaw-right” will arm the quad but the motors do not run at all.
Raising the throttle will get the motors to start running normally as expected.

Somehow the throttle-idle value is triggering a weird condition when arming with “yaw-right”.
If I arm after a normal GPS fix, everything works fine.
On the other hand, throttle-idle=18% is the minimum value I need to get the motor spin normally when arming.

Since airmode_type = THROTTLE_THRESHOLD and airmode_throttle_threshold = 1150 (by default), could it be that somehow airmode (?!?) kicks in at 18% throttle-idle?
I would like to be sure to solve this problem correctly as it can be very dangerous if one is arming with “yaw-right” and has propellers on…

## Expected behavior
Clearly I would expect to be able to arm with "yaw-right", for example when I am indoor and I do not have a gps fix, and get the motor to spin normally.

## Suggested solution(s)

## Additional context
CLI dump can be found under issue #9777

---

- FC Board name and vendor: SpeedybeeF405V3
- INAV version string: # INAV/SPEEDYBEEF405V3 6.1.1 Jun 13 2023 / 12:27:32 (42ced253)
# GCC-10.2.1 20201103 (release)

breadoven, 2024-03-12:
Idle throttle is:
`min_command + (throttle_idle * (max_throttle - min_command) / 100)`

So 18% idle = 1153 which is > `airmode_throttle_threshold ` of 1150 meaning Airmode becomes active as soon as you Arm causing a reaction to the Yaw right input.

17% idle is below this threshold so Airmode won't become active when you Arm with Yaw right. The issues with the unstable motor speed @17% idle and not running at all <17% are likely due to the idle calibration of the ESC given you're using Multishot protocol which needs calibrating/setting up as I understand it.

sgrizzi, 2024-03-12:
Thanks a lot!
Now I know what is happening and how to correct the problem.
As this can be a rather dangerous situation -the quad arms fine with a GPS fix but it goes mad with "yaw right"- maybe it could be a suggestion to include a warning when throttle-idle is set (inadvertently) > airmode_throttle_threshold.

I am planning to try Bluejay instead of BlHeli and see if this can also solve the problem. Then also move to INAV 7.
Thanks again

sgrizzi, 2024-03-17:
With the SpeedybeeF405V3 it was necessary to move to INAV 7 to be able to use Bluejay with Dshot300.
"throttle-idle" is now set at 7% so that arming with yaw-right does not get the motors to instantly spin at full throttle.

closed 2024-03-17 by sgrizzi