iNavFlight/inav #9791: Position Hold without GPS (dead-reckoning)

piotr-defendeye, 2024-03-12:
## Current Behavior

The Position Hold feature requires a magnetometer and GPS unit simultaneously. Without one of them, the feature is unavailable.

## Desired Behavior

The Position Hold could utilize the inertial sensors (IMU, Magnetometer) and Barometer to perform a dead-reckoning-based Position Hold. This would of course be much less accurate than the GPS-based Position Hold, however, for some applications - satisfactory.

## Suggested Solution

Utilizing the already available data from IMU+MAG+BARO and tracking the horizontal displacement over time with dead-reckoning algorithms. 

## Who does this impact? Who is this for?

For Quadcopter users.

## Additional context

Jetrell, 2024-03-12:
> The Position Hold feature requires a magnetometer and GPS unit simultaneously. Without one of them, the feature is unavailable.

INAV 7.1 RC1 allows a quad to still _navigate_ with magnetometer interference. Or if the magnetometer fails.. See release notes.

INAV will emergency land a MC, using the barometer, if a GPS fix is lost, or the GPS fails.
 I don't see much point it holding in a hover until the battery goes flat. Which is what PosHold with GPS dead reckoning would do.

piotr-defendeye, 2024-03-12:
> INAV will emergency land a MC, using the barometer, if a GPS fix is lost, or the GPS fails. I don't see much point it holding in a hover until the battery goes flat. Which is what PosHold with GPS dead reckoning would do.

My point is completely different. In indoor applications, or in GPS-jammed areas, the ability to **hover in place without the need for GPS** would be highly beneficial

MrD-RC, 2024-03-12:
> My point is completely different. In indoor applications, or in GPS-jammed areas, the ability to **hover in place without the need for GPS** would be highly beneficial

The INAV project doesn’t consider things like GPS-jammed area. INAV is not intended to be used in a war zone.

piotr-defendeye, 2024-03-12:
> The INAV project doesn’t consider things like GPS-jammed areas. INAV is not intended to be used in a war zone.

You don't need to be in a war zone to use a drone in a GPS jammed area. Unfortunately countries nearby active warzones suffer issues with GPS as well. Additionally, all you need to be is **indoor**

Jetrell, 2024-03-12:
> My point is completely different. In indoor applications, or in GPS-jammed areas, the ability to **hover in place without the need for GPS** would be highly beneficial

Use a Lidar/Optical flow sensor on your quad, in place of a GPS.

piotr-defendeye, 2024-03-13:
> > My point is completely different. In indoor applications, or in GPS-jammed areas, the ability to **hover in place without the need for GPS** would be highly beneficial
> 
> Use a Lidar/Optical flow sensor on your quad, in place of a GPS.

Lidar/Optical flow sensor is 1. additional cost 2. increase in BOM item count 3. additional assembly 4. additional build complexity 5. more wires

Inertial "stay in one place" feature would be enough for most of the use-cases in indoor use (where external factors such as strong winds are not present) and would not require any additional cost or complexity

MrD-RC, 2024-03-13:
By using the term BOM. It seems that this is not for indoor use. But for a large number of UAVs in GPS denied areas. Most indoor pilots are racing whoops, and don’t use POS HOLD.

piotr-defendeye, 2024-03-13:
> By using the term BOM. It seems that this is not for indoor use. But for a large number of UAVs in GPS denied areas. Most indoor pilots are racing whoops, and don’t use POS HOLD.

By using the term BOM it means when you do multiple camera angles for multiple locations 7 days a week you want to have more than one piece.

Why is the INAV community so negative?

MrD-RC, 2024-03-13:
BOM is Bill Of Materials. Used when making mass purchases. Your name is also defend eye. I’m not making a huge leap here. 

The INAV community is not negative. Out maintainer said since the start of the conflict that INAV is not made for war and no features will be added for that.

piotr-defendeye, 2024-03-13:
> BOM is Bill Of Materials. Used when making mass purchases. Your name is also defend eye. I’m not making a huge leap here.
> 
> The INAV community is not negative. Our maintainer said since the start of the conflict that INAV is not made for war and no features will be added for that.

BOM is Bill Of Materials. Also used when making many units of something, as in, multiple drones for your company to record indoor ads. Ardupilot already has dead reckoning return-to-home available, so it seems like a skill issue to me. 

Please tell me how the ability to remain still in one place indoors is considered a "war-related feature" to you? Also, your assumptions are incorrect and hurtful.

sp9dev, 2024-03-13:
What a roller-coaster here...

But yeah, it's safe to say, that such a feature would be highly useful. I cannot trully imagine it being something morally problematic, it's more likely your drone will be shot down when stationary anyways, so position-hold doesn't seem like a combat feature whatsoever. Would love to see IMU-based hover in INAV 👍

rts18, 2024-03-14:
@piotr-defendeye 
>and would not require any additional cost or complexity

Maybe not to you. How about the developers who have to write and test the code?

>Why is the INAV community so negative?

Maybe because they know iNAV and the complexity of implementing such a feature. And you don't. 

>Ardupilot already has dead reckoning return-to-home available

Then use ArduCopter.
Enlighten yourself by looking into how long it took Randy Mackay to write and test ArduCopter dead reckoning.

>so it seems like a skill issue to me

Then implement it yourself.
Don't expect any favors if you insult respected dev's with snide remarks like that. Then in the next comment, make a woke complaint like this.

> your assumptions are incorrect and hurtful.

Talk about entitlement!!

sensei-hacker, 2024-03-14:
So much passion about something that's pretty much not possible anyway. I have an accelerometer sitting here on the table. The total reading is varying from 0.994 to 1.01 while it's sitting  still on the table.

So let's estimate .003g error.

What is the result of an error of 0.003g acceleration for ten minutes?
Over a period of ten minutes, the expected error would be 17,304 feet! 5.2km error in ten minutes.

This would not be useful other than for a toy holding somewhat steady for a few seconds indoors. It would be useless outdoors, where the tilt needed to compensate for the breeze is indistinguishable from horizontal acceleration.

sensei-hacker, 2024-03-14:
So much passion about something that's pretty much not possible anyway. I have an accelerometer sitting here on the table. The total reading is varying from 0.994 to 1.01 while it's sitting still on the table.

So let's estimate .003g error.

What is the result of an error of 0.003g acceleration for ten minutes?
Over a period of ten minutes, the expected error would be 17,304 feet! 5.2km error in ten minutes.

This would not be useful other than for a toy holding somewhat steady for a few seconds indoors. It would be useless outdoors, where the tilt needed to compensate for the breeze is indistinguishable from horizontal acceleration.

If a person wanted to do outdoor navigation without GPS, for any reason , a ten cent IMU such as those on a typical flight controller simply isn't the way do that. It can be done optically and it can be done with a ring laser. The proposal here just plain won't work because math. Because double integration turns a tiny error in accelerometer measurement into a huge error in position.

closed 2024-03-14 by sensei-hacker

sensei-hacker, 2024-03-14:
Re Ardupilot, it's off by 800 meters in the course of two minutes - with an airspeed sensor.

In other tests, it gave up in 7 seconds.
[image]

sakis08, 2024-05-09:
any successful test getting poshold without gps or optical flow ?