iNavFlight/inav #9796: "Glass Panel" support

GTFreeFlyer, 2024-03-15:
## Current Behavior
Not implemented

## Desired Behavior
For the FPV pilots who want to put an LCD in the cockpit in front of the camera, would be neat to have "gauges" displayed on the LCD as an alternative to the OSD.  Maybe even take it a step further and set up a HUD display so that users can project it onto a beam splitter in front of the camera.  

## Suggested Solution
Take the data that is normally sent to the OSD, but allow it to be configured as a "steam gauge" or glass cockpit display for a small LCD display such as a 240x240 pixel approx 1 to 2" display.  Being able to plug the LCD directly into the FC's serial port (or whichever) would be preferable since all the data is already there.

## Who does this impact? Who is this for?
FPV pilots wanting more realism

Thanks for considering

b14ckyy, 2024-03-15:
Drawing a display from the FC won't work. Needs too much CPU time. All the OLED Displays are I2C that does not work as every display draw takes too long and interrupts the PID loop. A Serial Display would need a dedicated controller already and these systems already exist by using MSP or MAVLINK Telemetry to show data on small displays. 

So its basically already there.

GTFreeFlyer, 2024-03-16:
Thank you for the response.

closed 2024-03-16 by GTFreeFlyer