iNavFlight/inav #9801: Dive detection for Trackback RTG

DarthPlasma, 2024-03-17:
## Current Behavior
In case of a failsafe or switch-triggere RTH after or during a dive, the quad flyis back cutting through the obstacle trying to blast through an evetual obstacle present

## Desired Behavior
the quad should track back vertically up to the first point where the dive begins

## Suggested Solution #1
add a "dive detection feature"  deactivable that constist in three phases:

`RTH_Trackback_divedetection = true`

1) triggering the dive detection:
  conditions: pitch or roll down more than a preset angle `RTH_Trackback_divedetection_dive_angle` (default 50°, min 30 to 90) AND throttle below a preset value `RTH_Trackback_divedetection_dive_throttle `(default 20%, min = idle max = 40%)
                  OR
                throttle below  a preset value` RTH_Trackback_divedetection_drop_throttle` (default 15% max 30%)
                  OR
                throttle dips a predefined value in 1s (10ds) default = 30%  `RTH_Trackback_divedetection_drop_throttle_d`

a Trackback point is recorded immediately

2) first phase operation (first 5 trackback points)

take 5 points every default 150cm (adjustable from min 100 to 250 cm) `RTH_Trackback_divedetection_phase1_pointdistance`, vertical or horizontal, the largest among the two triggers the point recording

3) second phase operation (the rest of the points till the termination of the dive)
 
take a point every default 400cm (adjustable from 300 to 600 cm) `RTH_Trackback_divedetection_phase2_pointdistance`

dive termination condition:

vertical speed > -300 cm/s (adjustable from -200 to -600 cm) `RTH_Trackback_divedetection_termination_vario`
OR
throttle above 40% (adjustable from 30 to 60% cm) `RTH_Trackback_divedetection_termination_throttle`

after this revert to normal trackback logging

increase trackback points from 50 to 80

             

## Suggested Solution #2
Ad the possibility to choose to trigger phase 1 RTH (climb), this way using "EXTRA" will make the quad climb over the point of the dive, then switch to trackback RTH

## Who does this impact? Who is this for?
All cinematic mountainf surfers with close-to-peak flights

b14ckyy, 2024-03-20:
That's not a bug but a Waypoint navigation setting you can set. 
RTH trackback is basically a WP mission. 

Enable the Enforce Altitude at Waypoint option by setting a Altitude tolerance greater than 0 (0 disables). This will make sure, that Altitude has priority over position and the quad will only continue to the next WP in the chain after the current waypoint altitude is reached. If this is set to 0 (default) the horizontal position only will be used as a confirmation trigger and the altitude changes are cut short. 

this can be closed

breadoven, 2024-05-14:
Just noticed this. Reopening because Enforce Altitude actually only works for WP missions, it won't work for RTH Trackback.

DarthPlasma, 2024-05-15:
Maybe a simple Solution is to insert stage 1 failsafe (climb) bifore rth trackback