iNavFlight/inav #9806: Altitude hold need improvement

Lalbabu2001, 2024-03-18:
## Current Behavior

Altitude hold does strange things when throttle stick input is applied.
## Steps to Reproduce

1.arm quad enable althold 
2.lift up using throttle stick 
3.increase throttle and when centred quad sinks down couple of meters.
4.decrease throttle to lower from mid and when again centred quad raises couple meters.

## Expected behavior

As quad is directed to go up and when throttle centred it should stop at that position then why it sinks or vice versa.
## Suggested solution(s)

## Additional context

And also sometimes quad automatically starts to sink when in position hold. I have attached my blackbox log pic for now,  if necessary I will attach full log file.
---

- FC Board name and vendor:
- INAV version string: version 6.1.1

sensei-hacker, 2024-03-18:
Please post your diff. The behavior you are requesting is different from the default behavior, and would require appropriate settings.

P-I-Engineer, 2024-03-18:
this might be an icm gyro issue, what fc/gyro is it? try 7.1 rc1.

b14ckyy, 2024-03-20:
Can you please try this with 7.1, ideally with the latest build for 7.1 that I made yesterday: [https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl](https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8iOYA_eGtgz0T-w?e=A4mIfl&fbclid=IwAR2Iq7FSDmSOo5OWx5OeYTzEipYMGAq-Hbeeh3nizKgbhLF7JhmbfLeSF-M)

7.1 has optimizations in place to prevent altitude hops for multirotor.

b14ckyy, 2024-03-26:
Will close this for now as it should be much better in 7.1 already. You can still reply if you tested it.

closed 2024-03-26 by b14ckyy