iNavFlight/inav #9811: Add a NAME to identify each SafeHome GPS location

NorthIceMan, 2024-03-20:
## Current Behavior
# safehome
No idea where these safehomes are after a few days because they have no location NAME...
safehome 0 1 490180660 -664490499
safehome 1 0 492580550 -684690466
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0
## Desired Behavior
These safehomes have a location NAME...and are recognizable immediately...
safehome 0 1 490180660 -664490499 Pawel's Field
safehome 1 0 492580550 -684690466 Darren's Glenn
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0
## Suggested Solution
Add a NAME field to each SafeHome which could be entered in Mission Control (long enough to identify a location - how about 20 chars?)

## Who does this impact? Who is this for?
All users of iNAV 7

## Additional context
The current "no name" context can affect safety of how Min Distance to Safe Home Max Distance is set in Advanced Tuning

MrD-RC, 2024-03-21:
Personally. I don’t really feel that this is needed. When you load the safehomes in Mission Control, you can see where they are on the map.

NorthIceMan, 2024-03-21:
Thanks for the input Darren...
Yep-I was more thinking about a future ability of a "handle" to save steps (especially at the field) of selecting a new home position.  I'm not always in front of my laptop.  Maybe another way to do this is to replace the "handle" SafeHome with the name of preference? Then a new field would not be necessary.

b14ckyy, 2024-03-21:
Not sure what you describe exactly.
Safe Home Positions are permanently stored on the FC and edited in the configurator. 
If a Safe-Home location is within range of `safehome_max_distance`, it will be used by INAV when armed for Failsafe or RTH. You do not select them yourself. It will notifiy you in the OSD if a Safehome is active as soon as you arm.

NorthIceMan, 2024-03-23:
Thanks for the response.
I HOPE it is possible to select a SafeHome without letting iNAV auto-select
one.  It would extend the functionality and flexibility of using SafeHome's
greatly.
How about an ability to edit the word "SafeHome" and replace it with a
meaningful location name?

Ron

MrD-RC, 2024-03-23:
No, safehomes are not designed to work that way. It will always pick the nearest safehome to the launch point. It is intended as a safe landing area, especially useful in an emergency. The reason for the short maximum distance is to allow you to get to the safehome to make sure it is clear before your aircraft emergency lands.

NorthIceMan, 2024-03-23:
Ok, thanks for the clarification. Appreciate your response.

A little humour to brighten your saturday...
[image: image.png]

Ron

b14ckyy, 2024-03-23:
I guess topic is answered. Will close here now.

closed 2024-03-23 by b14ckyy

NorthIceMan, 2024-03-23:
Ok, thanks.