iNavFlight/inav #9816: Tmotorf7v2 has no accelerometer ICM42688 as an option

Delvien, 2024-03-22:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1. Flash current inav 7.0 target TMOTORF7v2 

## Expected behavior
Ability to select ICM42688  gyro

## Suggested solution(s)
Add driver for ICM42688 

## Additional context

INAV/TMOTORF7V2 7.0.0 DEC 5 2023 / 11:12:23 (895a4f31)
GCC-10.3.1 202210824 (release)

- FC Board name and vendor:T-motor F7 pro https://pyrodrone.com/products/t-motor-f7-pro-flight-controller-30x30mm
- INAV version string: 7.0.0

b14ckyy, 2024-03-22:
Not a bug. Use the ICM42605. They use the exact same driver and is also auto-selected when switching to auto. This is normal. 

[image]

Delvien, 2024-03-22:
AUTO does not automatically select ICM42605. When selecting ICM42605 it also does not work. 

[image]

[image]

Gyro in Betaflight works.

mmosca, 2024-03-22:
@b14ckyy I think gyro is missing from the target, probably because the manufacturer changed the board at some point due to availability.

Delvien, 2024-03-22:
> @b14ckyy I think gyro is missing from the target, probably because the manufacturer changed the board at some point due to availability.

I confirmed your build here: https://discord.com/channels/791437907478577172/881440703811559435/1220322311937327164

made gyro function.

b14ckyy, 2024-03-22:
Thx @mmosca 
these constant revision changes of some manufacturers are really a pain in the A

PR #9819 (2024-03-22): Add ICM42688P to TMOTORF7V2
Fixes #9816