iNavFlight/inav #9826: iflight blitz F722 V1.2 no ICM42688 support

AlexirN2, 2024-03-23:
## Current Behavior
Barometer And Accelerometer Not working

## Desired Behavior
They should work without problems

## Suggested Solution
Add support for ICM42688 on iflight blitz F722 v1.2

## Who does this impact? Who is this for?
Iflight nazgul5 v3 owners, quads with iflight blitz F722 v1.2 FC

b14ckyy, 2024-03-23:
Already fixed in https://github.com/iNavFlight/inav/pull/9797

Will be fine in inav 7.1. Release in about a week. Not fixed in RC1. If urgent, you can use the unofficial RC2. https://1drv.ms/f/s!Ar6QsGgoXRgqyPk8ttBS_Kp6Auy8xA

b14ckyy, 2024-04-12:
[inav_7.1.0_IFLIGHT_BLITZ_F722.zip](https://github.com/iNavFlight/inav/files/14961191/inav_7.1.0_IFLIGHT_BLITZ_F722.zip)

Please try this one. Also carefully try out all the UARTs and motor outputs if everything works correctly just in case more has changed on the 1.2 version.

Aroshasguy, 2024-04-12:
OK the accelerometer is functioning now. Thank you!
I will check the other stuff later.

b14ckyy, 2024-04-12:
I checked the iFlight page and that was the only change. So should be okay.
Btw just to make clear, barometer is working, if present, because you wrote barometer and Accelerometer in your initial report.

Aroshasguy, 2024-04-12:
Barometer not working. Not sure why.

b14ckyy, 2024-04-12:
ok gimme a second
I missed that part as I thought it was a typo.

b14ckyy, 2024-04-12:
[inav_7.1.0_IFLIGHT_BLITZ_F722.zip](https://github.com/iNavFlight/inav/files/14963484/inav_7.1.0_IFLIGHT_BLITZ_F722.zip)

Baro should work now with this one.

b14ckyy, 2024-04-12:
Strange thing is, the FC should have a DPS310 that was already defined. If this build works for you please tell me what baro is auto detected when setting the option to AUTO.

Aroshasguy, 2024-04-12:
Still not working. On both versions it goes back to NONE when I set to AUTO

b14ckyy, 2024-04-12:
what GPS do you use? @mmosca just told me that there is a GepRC GPS that also has an integrated DPS310 baro that could clash with the same address as the on board one. I just checked with the betaflight target and there is also a DPS310 defined, same as in INAV. I2C pin definition is also identical.

Aroshasguy, 2024-04-13:
I have iFlight M8Q-5883.
Just tried going back to Betaflight 4.4.3, and the barometer is not functioning there too.
Maybe I have a faulty barometer. I’ll contact iFlight to see what they say. Maybe someone else with the same FC could confirm.

b14ckyy, 2024-04-13:
That is a Magnetometer, not a barometer 😁

b14ckyy, 2024-04-13:
Can you please unplug your GPS and see if the Baro Ia then detected?

Aroshasguy, 2024-04-13:
After unplugging the GPS, the barometer was detected as DPS310 on both versions, and also on Betaflight 4.4.3. Do you think this conflict can be solved?

This is the GPS:
https://www.getfpv.com/iflight-m8q-5883-v2-0-gps-module-w-compass.html

b14ckyy, 2024-04-13:
I have seen this problem between DPS310 and QMC5883L a few times now when on the same I2C bus. 

do you have set your I2C speed to the recommended 800kHz? If yes try 400kHz and see if that helps to get detected.

Aroshasguy, 2024-04-13:
I have done some extensive testing.
It appears that everything is working correctly only when the battery is connected.
I can confirm that both versions work, on both 400Khz and 800Khz, and also on betaflight.
If I plug the battery before connecting USB, it works best.

Sorry for the hassle and thanks for your efforts.

One last question, which one of your versions should I use? The first or the second? What is the difference?

b14ckyy, 2024-04-13:
The first one is fine. The second one just adds all Baro drivers to the code but that was not the problem. It just wastes a bit of flash storage. So it does not matter wich one you use.

I created a PR to fix the gyro definition in 7.1.1 that comes soon.

closed 2024-04-13 by b14ckyy