iNavFlight/inav #9833: INAV 7.0.1/7.1.0RC1 Multirotor: ESC hiccups on BLHELI_32 4-IN-1 ESC (not present in INAV 6.1.1)

Pairan, 2024-03-23:
## Current Behavior
While hovering or in flight the ESCs seem to get a hiccup randomly. The used Stack is **MAMBA F55_128K**

This Behaviour is not present in INAV 6.1.1 which I just checked on this build. 

## Steps to Reproduce
Flash INAV 7.x and Fly

## Expected behavior
Clean motor controls, no hiccups 

## Additional context
I only got a DIFF /ALL on this. 
https://pastebin.com/Cb1KxESK

---
# version
# INAV/MAMBAH743_2022B 7.1.0 Feb 14 2024 / 15:16:47 (49325426)
# GCC-10.3.1 20210824 (release)
AND also 
# INAV/MAMBAH743_2022B 7.0.0 Dec  5 2023 / 10:26:54 (895a4f31)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-23:
Can you provide a blackbox log please? 
Also can you try to disable blackbox for one flight and check if you still experience hickups?

Pairan, 2024-03-23:
i will give it a try and report back later

Pairan, 2024-03-23:
Hiccups appear WITH and WITHOUT Blackbox.
[Uploading blackbox_log_7.1_2024-03-23_143611.TXT…]()

b14ckyy, 2024-03-23:
Can you please test this 7.1 Build and see if this is still an issue. This is for testing purposes only to narrow down the root cause. 
[inav_7.1.0_MAMBAH743.zip](https://github.com/iNavFlight/inav/files/14731842/inav_7.1.0_MAMBAH743.zip)

Pairan, 2024-03-23:
something is wrong with that firmware ... missing the GYRO

Arming disabled flags: ANGLE CAL NAV HWFAIL CLI DSHOTBEEPER
OSD: MSP DisplayPort [53 x 20]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 0/1
    Glonass 0/0
    Max concurrent: 3

Pairan, 2024-03-23:
or did you pick the wrong build target? => MAMBAH743_2022B

b14ckyy, 2024-03-23:
You are right. My bad. Didn't realize that both targets use the same ta
[inav_7.1.0_MAMBAH743_2022B.zip](https://github.com/iNavFlight/inav/files/14732113/inav_7.1.0_MAMBAH743_2022B.zip)
rget source file and Compiled only MAMBAH743. Sorry. Here is the correct one:

b14ckyy, 2024-03-23:
What I would like to know is, if the enabling of the second gyro, despite not being actively used but available for switching if needed, is causing any clash in the cpu to cause this. As this is the first time I see this issue pop up and only on that board so far, it makes sense to check that. 

If this does not change the symptoms or fixes it, then it must be something with the dhsot signal itself but that has to be analyzed by a more experienced and better equipped developer. 

@mmosca has the same FC and had a very very similar issue, but the filter changes in 7.1 have fixed it for him. So it might be worth to compare the setups.

Pairan, 2024-03-23:
Hiccups are one ... as for now I would assume it to be fixed - Well done and that's for your time on that.

b14ckyy, 2024-03-23:
Test it a bit more please and make sure its gone 100%. If that'Äs the case I check with pawel and mabye we revert this change for the final release.

b14ckyy, 2024-03-23:
btw one thing that MIGHT still be an option is, that one of your 2 ICM42688-P gyros is faulty. I can't tell wich one is used if both are active in the public released firmware. If the now disabled gyro (with my build) is also not used with both active but is faulty, it could casuse data line issues on SPI that cause the hiccups. I will check with the devs and maybe we can reproduce the issue on known good hardware.

mmosca, 2024-03-23:
> btw one thing that MIGHT still be an option is, that one of your 2 ICM42688-P gyros is faulty. I can't tell wich one is used if both are active in the public released firmware. If the now disabled [...]

The changes that fixed it for me should have been included in rc1, so if you try inav 7.1rc1 I would expect your hickups to be gone as well, unelss there is an issue with one of your gyros, and @b14ckyy firmware that uses the other one is what fixed it for you.

Inav 7.1 should be released soon. If it still happens once inav 7.1 is out, that points it to an issue with one of your gyros.

mmosca, 2024-03-24:
@Pairan if you can test the firmware for the above pr: https://github.com/iNavFlight/inav/actions/runs/8408966743#artifacts

MAMBAH743_2022B will use the same gyro as 6.x
MAMBAH743_2022B_GYRO2 will use same gyro as 7.0

Please recalibrate the gyro when using it for the first time, as each chip may have slightly different calibration values, just in case.

If both work, it is great and it may have been a calibration issue.

If only one works, it is likely a hw issue on one of the gyros (either damaged gyro, or interference that affects only of the 2 gyros)

b14ckyy, 2024-03-26:
I think we can consider this as solved, as a probably hardware fault. Will close now.

closed 2024-03-26 by b14ckyy