iNavFlight/inav #9847: GPS ON/OFF setting only switches nav-features but not telemetry/monitoring

b14ckyy, 2024-03-24:
## Current Behavior
`feature GPS` fully disables all GPS functions, including Telemetry and OSD Monitoring

## Desired Behavior
It would be better if `feature GPS` would only disable navigation features like RTH and Nav Modes but keeps OSD info for speed, altitude, position enabled. It should also keep core AHRS functions enabled like Wind and G-Force drift compensation. 

## Suggested Solution
Make the feature switch only disable any kind of navigation function but keep the rest of the GPS functionality enabled if turned off. If someone wants to fully disable all GPS features, this can be done in the Port settings like every other connected hardware. 

## Who does this impact? Who is this for?
many pilots just want to have the GPS for OSD Info but always fly fully manual. Scale pilots for example or Hotliners to measure speed records. Making sure that no accidental navigation mode is set on an untuned plane is extra work and sometimes risk. So the safe way is to disable GPS features but this also kills the speed measurements and G-Force compensation. Having a dedicated GPS switch is kind of redundant to disabling it in the Ports while a "No-Nav-Switch" makes more sense. 

## Additional context
-

mmosca, 2024-03-24:
Not sure on this one. Features are often turned off to reduce cpu usage. this would keep most of the gps cpu usage.

DzikuVx, 2024-03-24:
I'm not sure about the gain. If you don't enable GPS modes then you don't have them. Is it only about fix before takeoff?

b14ckyy, 2024-03-24:
The fix before takeoff is another thing indeed. 
The alternative could be to just remove that switch entirely. As it is redundant to the port enable I guess.

mmosca, 2024-03-26:
It is hard to find the balance here.
Even if you don't have navigation modes, there are still osd elements that may need a fix before arm to work. (Anything that references home on the osd: Distance to home, home arrow, radar, etc)

Bypassing lock to arm is fairly easy with stick commands today as well.